espressif/esp-idf #96: Floating division bug

danicampora, 2016-11-13:
Hello,

We've found a bug that seems to be caused by the compiler, and it's only related to floating point divisions, and only when the values to be divided are part of the parameters passed to a function. Please run the following test (using the idf application template) to reproduce it. Thanks!

```python
#include <stdio.h>

void do_first_test(void)
{
    printf("starting first test\n");
    printf("divide two consts: %f / %f = %f <-- this is the proper result\n", 100000000.0, 1.0, 100000000.0 / 1.0);
}

void do_second_test(void)
{
    float x, y, z;
    printf("\nstarting second test\n");
    x = 100000000.0;
    y = 1.0;
    z = x / y;
    printf("if the division terms are stored in variables inside the function, the result is %f\n", z);
}

void do_bug(float x, float y) {
    float z;

    printf("\nshowing the bug here!!!\n\n");
    printf("the variables are x = %f, y = %f\n", x, y);
    printf("lets be sure, is x == 100000000.0 ? %d, is y == 1.0 ? %d (1 means yes)\n", x == 100000000.0, y == 1.0);

    z = x / y;
    printf("obtained division result = %f\n", z);
    printf("result from performing the division straight from variables = %f\n", x / y);
    printf("expected : %f\n", 100000000.0 / 1.0);

    printf("\nthat was it, now just to be sure, lets do some other tests\n");
    printf("now the dividend is a const. Result is = %f\n", 100000000.0 / y);
    printf("now the divisor is a const. Result is = %f\n", x / 1.0);
}

void extra_bug_demo(float x, float y)
{
    printf("%f/%f = %f\n", x, y, x / y);
}

void app_main()
{
    printf("\n\n\nDivision bug demo:\n\n");
    do_first_test();
    do_second_test();
    do_bug(100000000.0, 1.0);
    printf("\nthis is another example of the bug: ");
    extra_bug_demo(80000000.0, 1000000.0);
    printf("but with other values, it doesn't fail: ");
    extra_bug_demo(100.0, 1.0);
}
```

Any idea what might be going on? Thanks!

Cheers,
Daniel

nkolban, 2016-11-13:
Howdy ... would it be possible to post the textual output of the reported problem.  While I know we can download the code, compile it, run it and then examine the results ... I can't help but feel it would be easier for many of us to see the output and see the results that are causing concern without going through those compilation steps.  We might get others who aren't as familiar with compilation techniques to see the story and go "aha!!".

igrr, 2016-11-14:
Hi Daniel,
The bug in the floating point division instruction sequ

igrr, 2016-11-14:
Hi Daniel,
There is a bug in the current version of the toolchain, introduced in this commit:
https://github.com/espressif/crosstool-NG/commit/8d95cadb60062eb6c2db82a8ac67eaa802dfb4fa

You may build the toolchain from source from https://github.com/jcmvbkbc/crosstool-NG, branch xtensa-1.22.x. Once we verify that all libraries work okay with new toolchain, we'll update precompiled toolchains and repository links.

danicampora, 2016-11-14:
@igrr thanks for the quick reply. We'll try with the rebuilt tool-chain, and report results asap.

Cheers,
Daniel

projectgus, 2016-12-22:
Hi @danicampora ,

Did the updated toolchain fix this problem?

Angus

danicampora, 2016-12-22:
Yes, it did :-)

projectgus, 2016-12-22:
Yay!

igrr, 2016-12-22:
Reopening — I still need to rebuild the toolchain for Windows and update download links.

igrr, 2017-01-16:
Toolchain links have been updated.

closed 2017-01-16 by igrr

cnlohr, 2017-03-09:
I know the issue is closed, but I am really wondering how floating point division is done on the ESP32.  Are there undocumented assembly opcodes for it or is it using some library?

igrr, 2017-03-09:
@cnlohr This is the implementation of fp division, sqrt, and reciprocals :
https://github.com/espressif/crosstool-NG/blob/HEAD/local-patches/gcc/5.2.0/0003-xtensa-add-HW-FPU-sequences-for-DIV-SQRT-RECIP-RSQRT.patch

cnlohr, 2017-03-09:
Oh my... Just making sure.  This all comes with the normal IDF, right? Also, I am perplexed by the code, considering it makes use of the ```div0.s```, ```rsqrt0.s```, ```const.s```, ```mkdadj.s``` and ```nexp01.s``` instructions, which as far as I can tell is... undocumented?  Did you guys add them specifically to the ESP32, or are they part of a as yet proprietary xtensa processor specification?

No, really, there are two google results for "mkdadj.s" and the same for "rsqrt0.s"

igrr, 2017-03-09:
Yes, this comes with xtensa-esp32-elf toolchain you install as part of ESP-IDF setup.

These instructions are not specific to the ESP32, they are part of Xtensa single precision FPU option. The reason you don't get Google results for them is that the leaked "isa_rm.pdf" you can find on the Internet is fairly outdated. The more up-to-date ISA reference has not been released by Cadence for public use.

cnlohr, 2017-03-09:
Aww maaaaan.  How am I gonna do crazy things with DSP that shouldn't be possible without the ISA?  Exhaustive opcode exploration, here I come.

I knew something was wrong when I shoved some of these instructions in and got opcodes that didn't show up /anywhere/.

```
fa42b0        	nexp01.s	f4, f2
```

Thanks for the info, though!  Will try to get everything working with what you've provided.

eldadwasserman, 2018-04-03:
Hi All,

I have the same bug - with newer toolchain... 

this code line crash the MCU.... 

AC_average_float = (float) AC_total_int / AC_loop_cycles_int;

What can i do ???