espressif/esp-idf #100: include file "uECC.h" missing

RobinCutshaw, 2016-11-15:
Just did a git pull and makes now fail with:

CC src/secure_boot_signatures.o
/home/robin/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:21:18: fatal error: uECC.h: No such file or directory
 #include "uECC.h"
                  ^
compilation terminated.

projectgus, 2016-11-15:
Hi Robin,

Thanks for reporting this, sorry for the hassle. You can run `git submodule update --init` to get past this.

The build system should check this submodule automatically, but isn't. I'll get a fix for that into the repository ASAP.

Angus

RobinCutshaw, 2016-11-15:
That worked, thanks.  There are a couple of warnings still:

/home/robin/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:49:14: warning: unused variable 'digest_len' [-Wunused-variable]
     uint32_t digest_len;
              ^
/home/robin/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:48:27: warning: unused variable 'digest_data' [-Wunused-variable]
     const uint8_t *data, *digest_data;
                           ^

closed 2016-11-18 by igrr

polineer, 2019-06-13:
Hello

I'm facing this issue but I can't get the remedy right. Can anyone help please ?
 here is my screen shot
spuremicro@UbuntuServer:~/esp/esp-idf/examples/get-started/blink$ make flash
Toolchain path: /home/spuremicro/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
/home/spuremicro/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:22:10: fatal error: uECC.h: No such file or directory
 #include "uECC.h"
          ^~~~~~~~
compilation terminated.

projectgus, 2019-06-14:
Hi @polineer,

What version of IDF are you using and how did you install it? You can change to the IDF directory and run `git describe --tags` to check.

Can you please confirm that running `git submodule update --init` in the IDF directory doesn't change anything?

polineer, 2019-06-14:
Hello projectgus

Thanks for your consideration once more on this closed issue.

I followed this guide : https://docs.espressif.com/projects/esp-idf/en/stable/get-started/

When I switch to IDF directory and run git submodule update --init , nothing happens - it behaves like hiiting the "enter" key.

git describe --tags returns  "v3.3-beta1"

iduquegit, 2019-06-18:
Hello,

Same problem here, I have updated the Toolchain and Compiler version to the latest and I'm still getting the same error with "v3.3-beta1"

dhrishi, 2019-06-18:
It looks like you have switched from master branch. On master, micro-ecc component has been moved in `components/bootloader` directory in recent times from its original location in `components`

On the branch you are working with, check the status `git status` and just remove the untracked folders (the new bootloader subcomponent should be one of them) and give it a try.

iduquegit, 2019-06-20:
Hi!
I was using the esp32-at firmware that uses a previous version of the IDF (while I was in master branch). I just cleaned the repository removing the untracked folders and it is working now. 
Thank you for your help @dhrishi :)

polineer, 2019-06-20:
Hello guys
Thanks for the quick turn around on this.

I have also done some cleaning and it looks good now.

Thank you all for your support.

Regards