espressif/esp-idf #101: Build system refactoring resulted in broken $(abspath)

jmattsson, 2016-11-16:
I just attempted to upgrade NodeMCU32 to the latest IDF, but the build is now failing for me. Upon closer inspection the build system has somehow mangled the path for a `-L` for the linker.

I have a custom linker fragment which needs to be added, which I do like so:
```
COMPONENT_ADD_LDFLAGS:=-L $(abspath ld) -T nodemcu_core.ld -lbase_nodemcu
```
yet now somehow, by the time the linker runs, turns that path into ` -L \/home/johny/src/DiUS/nodemcu-firmware/build/base_nodemcu/ld` rather than the expected ` -L /home/johny/src/DiUS/nodemcu-firmware/build/base_nodemcu/ld`- notice the prepended backslash. If I `$(warning ====$(abspath ld)====)` in the component.mk at that point, there is no backslash.

If I replace `$(abspath ld)` with `$(COMPONENT_PATH)/ld` or even `$(PROJECT_PATH)/components/base_nodemcu/ld` things work. I am at a loss to explain why `$(abspath ld)` is broken. It's weird enough that I don't feel good about just ignoring the issue...

projectgus, 2016-11-16:
There used to be an inconsistency with the current working directory: The project-wide variables (COMPONENT_ADD_LDFLAGS, COMPONENT_ADD_INCLUDEDIRS, COMPONENT_DEPENDS) were evaluated in component.mk with CWD set to `$(COMPONENT_PATH)`, whereas any other target/stage of component.mk was evaluated with CWD set to `$(BUILD_DIR_BASE)/$(COMPONENT_NAME)`

The refactoring unified these so that the make environment is the same for both kinds of component.mk evaluation, and the working directory is also now the same for both. We had `COMPONENT_ADD_LDFLAGS := ... $(abspath blah) ..` style stuff in our component.mk files as well, which needed changing to `$(COMPONENT_PATH)/blah`.

My reasoning is that making it more consistent (component.mk always evaluated in the same working directory) is preferable to having `$(abspath)` in the same file mean one thing in some contexts and another thing in other context. Sorry we didn't document this better for people with existing components, though..

Let me know if there's a reason this is sub-optimal for the way you're using esp-idf, and we'll figure something out.

closed 2016-11-16 by projectgus

jmattsson, 2016-11-16:
Oh I'm all in favour of that unification, it just took me a little while to discover `$(COMPONENT_PATH)`. What I didn't (and still don't) understand is where that leading backslash is introduced into the `$(abspath)` result (but nothing else). With me using direct (`:=`) assignment, there should be no late evaluation where it gets changed. It had the "smell" of something that might be indicative of a latent bug somewhere deeper, so I thought I'd report it.

projectgus, 2016-11-16:
Right, I had not skimmed over the backlash. That is weird...