espressif/esp-idf #106: system_deep_sleep works?

jolivepetrus, 2016-11-17:
I attach a test code, extracted from 06_sntp example. In previous comments we cloud enter  the CPU in sleep mode in a similar way as exposed in test code.

In test code CPU doesn't enter in deep sleep.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/597500/sdkconfig.txt)
[sntp_main.txt](https://github.com/espressif/esp-idf/files/597501/sntp_main.txt)

igrr, 2016-11-24:
@jolivepetrus, please try running the example with the latest code in master. `system_deep_sleep` has been rewritten and the example is working in my tests.

jolivepetrus, 2016-11-24:
Perfect, I confirm that with latest code in master system_deep_sleep works fine.

Thanks.

closed 2016-11-24 by jolivepetrus