espressif/esp-idf #108: 04_https_request Compiler Error

ksitko, 2016-11-18:
Example 04_https_request fails to compile once it reaches EMBED server_root_cert.pem.txt.o

```
C:\msys32\opt\xtensa-esp32-elf\bin\xtensa-esp32-elf-objcopy.exe: supported targets: elf32-xtensa-le elf32-xtensa-be elf32-littnary ihex
make[1]: *** [/c/esp-idf/make/component_wrapper.mk:209: server_root_cert.pem.txt.o] Error 1
make: *** [/c/esp-idf/make/project.mk:324: main-build] Error 2

```

I am able to make all other examples.

negativekelvin, 2016-11-18:
Works for me on win7 x64 prebuilt msys2 env

ggsubs, 2016-11-19:
change OBJCOPY_EMBED_ARGS in make/component_wrapper.mk
`OBJCOPY_EMBED_ARGS := -I binary -O elf32-xtensa-le --binary-architecture xtensa --rename-section .data=.rodata.embedded`

negativekelvin, 2016-11-20:
@ggsubs  related to

https://github.com/espressif/esp-idf/commit/12dd886ee194004ce8f9e74c1b3329b3b40129be ?

projectgus, 2016-11-20:
Hi @ksitko ,

Sorry we didn't update this earlier. I think we found this bug internally at about the same time you reported this issue. Seems some objcopy versions (OS X perhaps?) didn't like the ambiguous argument but Linux & Windows were OK with it, even when built with the same crosstool version.

Latest esp-idf master branch with 12dd886 should fix the issue, but please reopen if it doesn't work for you.

Angus

ksitko, 2016-11-20:
@projectgus 
I'm a little confused, I'm running this in W10, not OSX. Also @negativekelvin I am using a fresh enviorment following the alternative step 1 instructions. I would add that the shell script for windows needed some modification to add to PATH for python and create a /opt folder for the toolchain.

ksitko, 2016-11-20:
Okay, it compiled just fine after updating to the latest master branch. Thank you so much @projectgus

closed 2016-11-20 by ksitko

projectgus, 2016-11-20:
Sorry @ksitko, please disregard my previous reply.  I accidentally misread the error message as being the exact same bug that was resolved in 12dd886.

I'm glad the problem seems to have been resolved in any case, but please reopen if you think something is missing from the non-prebuilt version of objcopy.

I'll take a look at missing steps in the shell script as well, thanks for the heads-up.

ksitko, 2016-11-20:
@projectgus I should warn I only tried compiling, I have not yet tested the example to see it actually works. I'm currently traveling thus I think it would be rather impolite to set up my board and try flashing it right now. I'll try in a few hours once I am able to.

As for windows_install_prerequisites.sh:

I found adding 
`export PATH="$PATH:/mingw32/bin"  
`right before the python -m pip install command and making a /opt folder in the msys32 folder allowed for the installation to complete correctly.

ksitko, 2016-11-21:
@projectgus okay, I have confirmed the https example runs now for me.