espressif/esp-idf #109: UART noise sensitivity leads to crashes

ggsubs, 2016-11-19:
With UART1 enabled and no hardware flow control the the OS periodically crashes when a 3" open wire is on the RX line. In between crashes, frequent UART_BREAK events received. In some cases of the crash corrupts the flash resulting in "E (538) boot: nothing to load". 
Pulling down the RX line w 10K helps to eliminate the "open wire RX" instability.

ggsubs, 2016-11-19:
The crash is preventable by disabling the framing error and break detection interrupts, either in uart.c or from the application code:

```
uart_driver_install(uart_num, 1024 * 2, 1024*4, 10, UART_INTR_NUM, &uart1_queue);
ESP_INTR_DISABLE(UART_INTR_NUM);
#define UART_EMPTY_THRESH_DEFAULT  (10)
#define UART_FULL_THRESH_DEFAULT  (120)
#define UART_TOUT_THRESH_DEFAULT   (10)
uart_intr_config_t uart_intr = {
  .intr_enable_mask = UART_RXFIFO_FULL_INT_ENA_M
  | UART_RXFIFO_TOUT_INT_ENA_M
  //| UART_FRM_ERR_INT_ENA_M
  | UART_RXFIFO_OVF_INT_ENA_M,
  //| UART_BRK_DET_INT_ENA_M,
  .rxfifo_full_thresh = UART_FULL_THRESH_DEFAULT,
  .rx_timeout_thresh = UART_TOUT_THRESH_DEFAULT,
  .txfifo_empty_intr_thresh = UART_EMPTY_THRESH_DEFAULT
};
uart_intr_config(uart_num, &uart_intr);
ESP_INTR_ENABLE(UART_INTR_NUM);
```

projectgus, 2016-12-22:
Thanks for reporting this. The wrong interrupt type was being cleared in the UART driver ISR, which lead to continual interrupts and eventual interrupt watchdog triggering.

Fix should be merged very soon.

projectgus, 2016-12-30:
Fix merged in 8bcd341fcae23c534ea98f47d1ba33e1694a9e48

closed 2016-12-30 by projectgus