espressif/esp-idf #112: 04_https_request certificate format incorrect

ggsubs, 2016-11-19:
The server cert file included in the example is in an incorrect format. The file is pem, the example expects der encoded. The cert file can be converted by:

```openssl x509 -inform pem -in server_root_cert.pem -outform der -out server_root_cert.der  ```

negativekelvin, 2016-11-20:
Example works for me as is.

projectgus, 2016-11-20:
What caused you to think that the example code expects a DER formatted certificate?

The certificate data is passed to the mbedTLS function `mbedtls_x509_crt_parse`, which supports both PEM and DER format certificates provided that support is enabled in the mbedTLS configuration file. We should have support for both in the default `esp_config.h` configuration in esp-idf.

DER format certificates are a computationally simpler to parse (no decoding step), but PEM is a lot easier to work with which is why I chose it for the example.

ggsubs, 2016-11-20:
When the pem format used mbedTls returns a format error:
```
I (396) example: Seeding the random number generator
I (406) example: Loading the CA root certificate...
E (406) example: mbedtls_x509_crt_parse returned -0x2180
```
The example only worked when replaced the cert with der format.

projectgus, 2016-11-21:
Thanks for clarifying. I don't get this failure on the current esp-idf master 8ab4e11:

```
I (207) example: Seeding the random number generator
I (207) example: Loading the CA root certificate...
I (217) example: Setting hostname for TLS session...
I (217) example: Setting up the SSL/TLS structure...
```

Maybe a clean build might help, or there's some other modification or config tweak which will allow me to reproduce?

Angus

igrr, 2016-11-22:
We have reproduced this on OS X. @ggsubs which OS are you running?

igrr, 2016-11-24:
Fixed in 5ac7810.

closed 2016-11-24 by igrr