espressif/esp-idf #114: Sending a datagram message to a co-routine

nkolban, 2016-11-21:
This one is going to take a little explaining.  I am working with a 3rd party open source project (Mongoose).  That uses an interesting signaling technique.

Imagine we have two sockets that are connected to each other within ESP32.  In principle, if we write on one, we can read on the other.  Now imagine two FreeRTOS tasks... call one "reader" and the other "writer"

The reader does:
```
void reader() {
   recv(sockReader, buffer, sizeofbuffer, 0);
}
```

while the writer does:
```
void writer() {
   send(sockWriter, buffer, size, 0);
}
```

Again, these execute on distinct tasks and we have ensured that reader() is running before writer().

When we run the code, we find that the writer() sends the request but reader() doesn't wake up/un-block.  It is as though the message is not delivered.

The sockets are connect to each other with the following initialization:
```
 sockReader = sockWriter = -1;

  (void) memset(&sa, 0, sizeof(sa));
  sa.sin_family = AF_INET;
  sa.sin_port = htons(0);
  inet_pton(AF_INET, "127.0.0.1", &sa.sin_addr.s_addr);

  sockReader = socket(AF_INET, SOCK_DGRAM, 0);
  rc = bind(sockReader, (struct sockaddr *)&sa, len);
  rc = getsockname(sockReader, (struct sockaddr *)&sa, &len);
  sockWriter = socket(AF_INET, SOCK_DGRAM, 0);
  rc = connect(sockWriter, (struct sockaddr *)&sa, len);
  rc = getsockname(sockWriter, (struct sockaddr *)&sa, &len);
  rc = connect(sockReader, (struct sockaddr *)&sa, len); 
```

The real world code snippet within Mongoose can be found here:

https://github.com/cesanta/mongoose/blob/master/mongoose.c#L3706

Attached to this issue is my full main.c that I used to test.

[main_c.txt](https://github.com/espressif/esp-idf/files/605082/main_c.txt)

negativekelvin, 2016-11-22:
http://www.nongnu.org/lwip/2_0_0/group__lwip__opts__loop.html

nkolban, 2016-11-22:
That's a good thought ... thank you kind sir.  I tried compiling lwip by adding:

```
CFLAGS += -DLWIP_NETIF_LOOPBACK=1
```

But that seemed to cause an almost immediate crash.  However I do think you are on to something.  If we didn't have a 127.0.0.1 address, that would produce a story similar to the experienced results in the initial problem.

negativekelvin, 2016-11-22:
I think you need at least LWIP_LOOPBACK_MAX_PBUFS and possibly other configuration

nkolban, 2016-11-22:
I think that puts me out of my depth ... hopefully someone with LWIP skills can assist.

negativekelvin, 2016-11-22:
Well if you look here those two might be enough:

https://github.com/espressif/esp-idf/blob/d467eabadbc90aa0703d3e131e5751f9d09ebdf8/components/lwip/include/lwip/port/lwipopts.h#L330

nkolban, 2016-11-22:
I tried setting those two by adding the following into my `component.mk`:

```
CFLAGS += -Wno-address -Wno-unused-variable -Wno-unused-but-set-variable -DLWIP_NETIF_LOOPBACK=1 -DLWIP_LOOPBACK_MAX_PBUFS=8
```
For the `lwip` component.  Unfortunately it still resulted in a crash at startup as opposed to when I remove them and no crash.  I don't disagree at all that these may be part of the solution, and hopefully someone will get to this issue to comment on it who has internals knowledge and experience.

nathanjel, 2017-01-25:
Solution that worked for me, this needs to be added into the main project Makefile

```
## LWIP settings
CFLAGS += -DLWIP_HAVE_LOOPIF=0
CFLAGS += -DLWIP_NETIF_LOOPBACK=1
CFLAGS += -DLWIP_NETIF_LOOPBACK_MULTITHREADING=1
```

This actually enabled me to do
- bind to 0.0.0.0 
- connect to self using device local IP address

nkolban, 2017-02-07:
@nathanjel Howdy ... I tried setting these values in my `component.mk` file for the lwip component ... unfortunately it resulted in a panic crash during startup.

nathanjel, 2017-02-07:
@nkolban As far as I am concerned it should not go to `component.mk` for LWIP. I have it in the main `Makefile ` in my project's root folder. My esp-idf is left unmodified. Could You try the same and then re-build Your project.

nkolban, 2017-02-07:
@nathanjel 
Thank you kind sir.  Your original instructions were clear and I had messed up.  I changed my main component.mk and all now compiles and runs correctly.  I haven't yet had the opportunity to test the loopback logic ... but so far so good.  Thank you again.

nkolban, 2017-02-07:
@nathanjel One more thought ... if I may ... in your post, you bound to 0.0.0.0 and the devices local IP address ... should I read this to mean that you did NOT test with 127.0.0.1?

nathanjel, 2017-02-18:
@nkolban 
This is how i bind: 
```
char * s_address = "0.0.0.0:1883";
netfsm->mqtt_server_cn = mg_bind(&mgr, s_address, mg_mqtt_broker);
```
and then this is how I connect - first i get the local IP from the event, and later i use it to connect locally
```
esp_err_t network_event_handler(void *ctx, system_event_t *event)
{
	switch (event->event_id) {
	case SYSTEM_EVENT_STA_GOT_IP:
		MN_Status.myipinfo = event->event_info.got_ip.ip_info;
```
```
sprintf(s_address, "%s:%d", inet_ntoa(localmq),
		        MQTT_INSECURE_PORT_NUMBER);
cptr = mg_connect_opt(&mgr, s_address, mqtt_client_event_handler, opts);
```

I tried multiple approaches before, don't remember exactly how 127.0.0.1 did behave

young-mu, 2017-02-28:
Is LOOPBACK available now?  After reading all the comments above, I am messed up. Who can **make a summary about how to use this feature**. Thanks.

nathanjel, 2017-03-02:
@young-mu it works for me very well as per the settings and code above (using Moongose). I have not tried it with lwip sockets directly.

liuzfesp, 2018-01-15:
@nkolban @young-mu @negativekelvin this is discussion long time before. In v3.0 or latest IDF, the option for enable/disable loopback was already added into menuconfig. The default configuration is enabled loopback as:

> #define LWIP_NETIF_LOOPBACK             1
> #LWIP_LOOPBACK_MAX_PBUFS              8

nkolban, 2018-01-15:
Thank you sir.  I think you are saying that loopback support is now in the latest available code base.  That is great news.  I'll close the issue.

closed 2018-01-15 by nkolban

snahmad, 2018-08-17:
I am getting compiler error.
error: implicit declaration of function 'mg_broadcast' on esp32.
on windows platform it build. but mg_broadcast does do nothing.

snahmad, 2018-08-17:
```
namespace
{
	static void ev_handler(struct mg_connection *nc, int ev, void *ev_data) 
	{

		switch (ev) {
		case MG_EV_POLL:
			break;
		case MG_EV_SEND:
			break;
		case MG_EV_ACCEPT:
			break;
		case MG_EV_CONNECT:
			break;
		case MG_EV_RECV:
			break;
		case MG_EV_CLOSE:
			break;
		default:
			break;
		}
	}

}

void sendUDPMessages()
{
	ACDTRACEF(ES_NoError, "Enter sendUDPMessages \n");
	
	struct mg_mgr mgr;
	mg_mgr_init(&mgr, NULL);

	struct mg_connect_opts opts;

	memset(&opts, 0, sizeof(opts));
	opts.flags = MG_F_ENABLE_BROADCAST;
	opts.iface = nullptr;

	
	const char *port1 = "udp://1457", *port2 = "udp://127.0.0.1:1457";
	const char *port3 = "udp://255.255.255.255:1457";

	struct mg_connection * nc_client = mg_connect_opt(&mgr, port3, ev_handler, opts);	
	
	bool stop = false;
	std::thread thread = std::thread([&]() {
		while (!stop) {
			mg_mgr_poll(&mgr, 1);
		}
	});
	
	for (int i = 1; i <= 10; i++)
	{
		std::string  send_msg = "hello";
		mg_broadcast(&mgr, ev_handler, (void*)send_msg.c_str(), send_msg.size());
	}
	
	std::this_thread::sleep_for(std::chrono::seconds(3));
	
	stop = true;
	if (thread.joinable()) {
		thread.join();
	}

	mg_mgr_free(&mgr);
	
}

```

Here is my example code.

ddomnik, 2024-02-06:
I have created a simple example. It may be helpful for others visiting this thread.
https://github.com/ddomnik/ESP-localhost-loopback-example/tree/main