espressif/esp-idf #116: Example  "06_sntp" does not wake up from deep sleep

krzychb, 2016-11-23:
When testing example 06_sntp on module "ESP32_Core_Board_V2" I have noticed issue with module not being able to wake up from deep sleep after first soft restart:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:0
load:0x3ffc0008,len:1964
load:0x40078000,len:3648
ho 0 tail 12 room 4
load:0x40080000,len:256
entry 0x40080034
I (555) heap_alloc_caps: Initializing heap allocator:
I (555) heap_alloc_caps: Region 19: 3FFC16C8 len 0001E938 tag 0
I (556) heap_alloc_caps: Region 25: 3FFE8000 len 00018000 tag 1
I (566) cpu_start: Pro cpu up.
I (572) cpu_start: Starting app cpu, entry point is 0x40080bf8
I (0) cpu_start: App cpu up.
I (587) cpu_start: Pro cpu start user code
rtc v134 Oct 20 2016 12:36:18
XTAL 40M
I (810) phy: phy_version: 246, Nov 18 2016, 17:30:07, 0, 0

I (1185) cpu_start: Starting scheduler on PRO CPU.
I (1186) cpu_start: Starting scheduler on APP CPU.
I (1186) example: Boot count: 1
I (1186) example: Time is not set yet. Connecting to WiFi and getting time over NTP.
tcpip_task_hdlxxx : 3ffc61a8, prio:18,stack:2048
frc2_timer_task_hdl:3ffc7c0c, prio:22, stack:2048
I (1216) wifi: pp_task_hdl : 3ffca46c, prio:23, stack:8192
I (1216) example: Setting WiFi configuration SSID sensor-net...
mode : sta(18:fe:34:6a:96:72)
I (2836) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:1
I (3676) wifi: state: 0 -> 2 (b0)
I (4676) wifi: state: 2 -> 0 (2)
I (5396) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (5396) wifi: state: 0 -> 2 (b0)
I (5406) wifi: state: 2 -> 3 (0)
I (5406) wifi: state: 3 -> 5 (10)
I (5456) wifi: connected with twc-net-3, channel 11
I (9196) event: ip: 192.168.1.116, mask: 255.255.255.0, gw: 192.168.1.234
I (9196) example: Initializing SNTP
I (9196) example: Waiting for system time to be set... (1/10)
I (11196) example: The current date/time in New York is: Wed Nov 23 14:18:28 2016
I (11196) example: The current date/time in Shanghai is: Thu Nov 24 03:18:28 2016
I (11196) example: Entering deep sleep for 10 seconds
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:0
load:0x3ffc0008,len:1964
load:0x40078000,len:3648
ho 0 tail 12 room 4
load:0x40080000,len:256
entry 0x40080034
I (553) heap_alloc_caps: Initializing heap allocator:
I (554) heap_alloc_caps: Region 19: 3FFC16C8 len 0001E938 tag 0
I (555) heap_alloc_caps: Region 25: 3FFE8000 len 00018000 tag 1
I (565) cpu_start: Pro cpu up.
I (571) cpu_start: Starting app cpu, entry point is 0x40080bf8
I (0) cpu_start: App cpu up.
I (586) cpu_start: Pro cpu start user code
rtc v134 Oct 20 2016 12:36:18
XTAL 40M
I (703) phy: phy_version: 246, Nov 18 2016, 17:30:07, 0, 1

I (705) cpu_start: Starting scheduler on PRO CPU.
I (706) cpu_start: Starting scheduler on APP CPU.
I (706) example: Boot count: 2
I (706) example: The current date/time in New York is: Wed Nov 23 14:18:38 2016
I (716) example: The current date/time in Shanghai is: Thu Nov 24 03:18:38 2016
I (726) example: Entering deep sleep for 10 seconds
```

After hard reset the situation repeats.

What could be the cause of this issue?

projectgus, 2016-12-22:
Was this issue resolved by the fix in #117 or some of the other recent changes to deep sleep behaviour?

krzychb, 2016-12-22:
Yes, the issue is resolved with current esp-idf.
Thank you for purging up this one.

closed 2016-12-22 by krzychb