espressif/esp-idf #118: missing `bootloader.map`

MobiusHorizons, 2016-11-24:
Updated the esp_idf today, and all the examples fail to compile. The relavant error message is 

``` sh
/Users/username/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/bin/ld: cannot open map file /Users/username/esp/esp-idf/examples/02_blink/build/bootloader/bootloader.map: No such file or directory
[/Users/username/esp/esp-idf/examples/02_blink/build/bootloader/bootloader.elf] Error 1

```

Where does this file come from, and how do I create it?

I think the documentation as well as the examples/template project should be updated to reflect this change.

igrr, 2016-11-24:
Just did a clean build of example project on macOS (which I think you are using) and it doesn't produce this error. Could you please run `make V=1` and post a link to the full output you get?

Also try checking of `build/bootloader` directory actually exists, by doing `ls -l build`. It should be created automatically by the build system, but the error you are getting may be indicating that it wasn't created for some reason.

MobiusHorizons, 2016-11-24:
You are correct, I am on OSX.

In my case, the `build` directory does not contain a `bootloader` subdirectory. 

The results of `make V=1` in the `02_blink` directory are as follows.

```
including /Users/paulmartin/esp/esp-idf/components/bootloader/Makefile.projbuild...
including /Users/paulmartin/esp/esp-idf/components/esp32/Makefile.projbuild...
including /Users/paulmartin/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including /Users/paulmartin/esp/esp-idf/components/mbedtls/Makefile.projbuild...
including /Users/paulmartin/esp/esp-idf/components/nghttp/Makefile.projbuild...
including /Users/paulmartin/esp/esp-idf/components/partition_table/Makefile.projbuild...
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/paulmartin/esp/esp-idf/components/bootloader/src V=1 BUILD_DIR_BASE=/Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader /Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/bootloader.bin
including /Users/paulmartin/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including /Users/paulmartin/esp/esp-idf/components/bootloader/Makefile.projbuild...
xtensa-esp32-elf-gcc -nostdlib -L~/esp/esp-idf/lib -L~/esp/esp-idf/ld -L/Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/esptool_py -L/Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/bootloader -L/Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/bootloader_support -L/Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/log -L/Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/spi_flash -L/Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/micro-ecc -L/Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/main -u call_user_start_cpu0  -Wl,--gc-sections -Wl,-static -Wl,--start-group  -lgcc -Wl,--end-group -Wl,-EL -o /Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/bootloader.elf -Wl,-Map=/Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/bootloader.map
/Users/paulmartin/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/bin/ld: cannot open map file /Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/bootloader.map: No such file or directory
collect2: error: ld returned 1 exit status
make[1]: *** [/Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/bootloader.elf] Error 1
make: *** [/Users/paulmartin/esp/esp-idf/examples/02_blink/build/bootloader/bootloader.bin] Error 2
```

I may have updated incorrectly, All I did was a `git pull; git submodule update`.

bflash4k, 2016-12-09:
I can confirm the build/bootloader is not created. If i create it manually, everything works.
My OS: Ubuntu 16.04 64b

bflash4k, 2016-12-10:
Update: not a bug. Please check the IDF_PATH var... you must set au full/complete path (without any ~)
eg: BAD: export IDF_PATH="~/Devel/Espressif/ESP32/20161209/esp-idf" 
 GOOD: export IDF_PATH="/home/bflash4k/Devel/Espressif/ESP32/20161209/esp-idf" 
Hope this will help !!!

MobiusHorizons, 2016-12-15:
@bflash4k This solution works for me. Thanks.

projectgus, 2016-12-16:
For the record, this also works:
```
export IDF_PATH=~/Devel/Espressif/ESP32/20161209/esp-idf
```

(Leaving off the quotes means that the shell expands the `~`, instead of storing it in the environment variable.)

I've submitted a fix for code review which will work around this issue entirely, so both types (quoted & unquoted) will work.

closed 2017-01-16 by igrr

projectgus, 2019-02-07:
@geekskick Are you referring to the "Linux and MacOS" section here: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/add-idf_path-to-profile.html#linux-and-macos

This works:

> export IDF_PATH=~/esp/esp-idf

Because the `~` is expanded by the shell before it processes the `export` command. You can verify by running `echo "$IDF_PATH"` to see the value.

A form of the command which does not work is this:

> export IDF_PATH="~/esp/esp-idf"

The quotes mean that shell expansion does not happen and the variable is set to the exact string value, including the `~`.

projectgus, 2019-02-07:
@geekskick Also, the fix linked above means that even if the IDF_PATH contains a literal `~` character, it's expanded properly. I just tested on the master branch and was able to build an ESP-IDF example with such a path set.

If the build is failing and you think it's a build system bug, please open a new issue with all of the information asked for in the issue template. Feel free to link back to this issue if you think it's related.

geekskick, 2019-02-07:
My apologies, I had the quotes in there! RTFM.

Although the second part doesn't seem to be true for me - with the ~ literal in the expansion doesn't work.

projectgus, 2019-02-08:
@geekskick What OS are you using?

geekskick, 2019-02-08:
macOS Mojave 10.14.2