espressif/esp-idf #120: Correct location of partition table in flash ...

nkolban, 2016-11-25:
I have a gut feel that the partition table is located in flash at flash address 0x8000.  This seems to contradict the address 0x4000 that is listed in the documentation as of the date of this post ... that article can be found here.

http://esp-idf.readthedocs.io/en/latest/partition-tables.html

projectgus, 2016-11-25:
Thanks @nkolban . We moved the partition table a couple of weeks ago and missed this documentation point.

There's an update in the pipeline for this.

nkolban, 2016-12-02:
I also seem to find that the old address of 0x4000 is documented in the ESP32-DevKitC getting started guide ... see ... https://espressif.com/sites/default/files/documentation/esp32-devkitc_getting_started_guide_en.pdf

projectgus, 2016-12-22:
Thanks for reporting that and sorry for the slow turnaround. I've relayed this to the documentation team, the Getting Started guide should be updated soon.

closed 2016-12-22 by projectgus