espressif/esp-idf #121: Request: API to determine NVS size and utilization ...

nkolban, 2016-11-25:
The amount of storage set aside for the NVS component is defined at flash time.  This issue requests an additional API to run-time query the amount of NVS storage space in use and free.

The background to this request is that NVS seems to be a simple enough mechanism to be used as a poor man's storage system when a file system might not be needed.  However, the desire to know how much space is in use would be extremely useful.

forthlightning, 2016-11-28:
I agree, I need to store a stream of sensor readings. The file system is a bit overkill because I basically need a giant buffer. Thanks!

igrr, 2016-11-29:
@forthlightning stream data is not something NVS is designed for. It's a key-value storage. If you need a giant buffer, take a look at partition API. You can add a partition to the partition table and then write to the partition at runtime.

nkolban, 2016-11-29:
There was also some talk about a file system technology to be released as part of ESP-IDF 1.0.  I'm not sure if that is still in plan or what kind of file system it may be, but this might also give you some form of balancing or wear leveling that might be important if you are writing lots and frequently.  I'm not sure what the technology specs say these days, but there were only a finite amount of writes one could perform against a specific flash set of cells.  So if you pick a specific flash area and keep over-writing it all the time, there may be problems.