espressif/esp-idf #122: example 05_ble_adv crashes

aircable, 2016-11-26:
I was not able to build this example so it actually works, it crashes after the first printf
Looks like download is ok but the API_vhci_host_check_send_available() crashes.
Any suggestions? No stack? Oh, the MAC address is zero, BT config is empty.
Is it not working yet?

rst:0x8 (TG1WDT_SYS_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:0
load:0x3ffc0008,len:1964
load:0x40078000,len:3648
ho 0 tail 12 room 4
load:0x40080000,len:256
entry 0x40080034
.[0;32mI (736) heap_alloc_caps: Initializing heap allocator:.[0m
.[0;32mI (737) heap_alloc_caps: Region 19: 3FFC5790 len 0001A870 tag 0.[0m
.[0;32mI (746) heap_alloc_caps: Region 25: 3FFE8000 len 00018000 tag 1.[0m
.[0;32mI (807) cpu_start: Pro cpu up..[0m
.[0;32mI (842) cpu_start: Starting app cpu, entry point is 0x40080acc.[0m
.[0;32mI (0) cpu_start: App cpu up..[0m
.[0;32mI (935) cpu_start: Pro cpu start user code.[0m
rtc v118 Oct 19 2016 15:22:11
XTAL 40M
.[0;32mI (1118) cpu_start: Starting scheduler on PRO CPU..[0m
BTDM CONTROLLER VERSION: 010102
.[0;32mI (217) cpu_start: StartBTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:00:3A:76
NVDS MAGIC FAILED
ing scheduler on APP CPU..[0m
phy_version: 187, Oct 10 2016, 19:23:46, 0
RF Init OK
BLE advt task start
ets Jun  8 2016 00:22:57

projectgus, 2016-12-22:
Hi @aircable,

Sorry for the delayed reply. I'm not able to reproduce this unusual crash (usually there is some crash dump info, whereas the board seems to be spontaneously resetting.)

What ESP32 hardware do you have? Is it possible there is a power supply problem causing a brownout when the radio transmits?

Angus

aircable, 2016-12-27:
Thanks for your reply. Updated to the latest version, the app works just fine.
I have this dev board: https://www.adafruit.com/products/3269

closed 2016-12-27 by aircable

preetampds, 2017-03-08:
Hi,
I am getting NVDS_MAGIC_FAILED in the latest version.