espressif/esp-idf #129: Enhancement request: Iteration over NVS namespaces, keys and types

nkolban, 2016-11-29:
This is an enhancement request for the NVS functional areas.

I would like the ability to have an API exposed that I can call that will return either an iteration or a list of the namespaces for each of the NVS sections.

Next I would like the ability to supply a namespace and have provided the list of keys and their types associated with that namespace.

The back story on this is as follows:

I was thinking over usage of NVS and it occurred to me that this is the place where application settings are saved.  However, it also appears to me that the NVS is not necessarily over-ridden when we flash a new version of the application.  This then opens up the possibility of having an application that could be used for debugging or maintenance or testing that would be a "loose" equivalent of the Windows "regedit" tool.  This would allow us to visualize the state of NVS data and either make changes or see what is present.  This tool could be written once and simply "interrogate" what is present and provide visualization and editing.  But in order to do that, we would need to expose some APIs to allow us to list the data contained within.

nkolban, 2017-02-09:
I had a use case come up where this capability would have been apparently necessary.  The use case was code that uses "environment" variables.   The thinking was that NVS would be used to store the name/value pairs of environment variables in permanent storage.  When the ESP32 boots, it would iterate through an "environment" namespace and for each name/value pair found, call `setenv()` to create an environment variable for access within C code.

StefanRvO, 2017-07-06:
I would be very interested in this too.
I am currently looking at using the NVS storage to store user information (user type, hash and salt). However, i would like the possibility of listing all existing users . For this to be possible, i would either need to store the list of users somewhere else, or be able to list all entries in the namespace i use for users.

Is this a feature which are being considered?

mringwal, 2017-10-18:
Here's another use case: for BTstack, I wanted to add support for storing which Characteristics have been subscribed by bonded device.

This could be done either as a single per-device blob or as a individual { device, characteristic }. The latter has the benefit that only a small (e.g. 4 byte entry) needs to be updated instead of the whole blob on any change.

The work around without the iterator would be to use some prefix + an index to get all entries. With the hash list used by nvs_flash, the overhead is not horrible, but an iterator would be more elegant/efficient in most or similar cases.

Edzelf, 2017-12-04:
I fully agree.  That's why I made a demo (Arduino sketch for the ESP32) to list all keys in the NVS.  See my github pages.

mringwal, 2017-12-04:
@Edzelf thanks! I'll have a look.

Koulwa, 2018-06-11:
any progress on this? its been a few months, wondering if espressif has mentioned any sort of inclusion of iteration.

berkutta, 2019-03-29:
+1 from my side. There are many use cases for NVS which need a method for listing all keys.

themadsens, 2019-04-08:
+1 from here. It is pretty hopeless to poke at all possible keys to see whether they exist.

themadsens, 2019-04-09:
Update:
Thanks for that code @Edzelf. I have repackaged it as an iterator [in this gist](https://gist.github.com/themadsens/026d38f432727567c3c456fb4396621b) for anyone interested

Edzelf, 2019-04-09:
That's the spirit!

closed 2019-06-12 by igrr

mringwal, 2019-06-12:
Thanks a bunch! I can add an iterator to BTstack's TLV storage as well (which is just mapped to esp-idf NVS on ESP32).