espressif/esp-idf #130: Doc: Cannot find function “nvs_flash_init_custom”

krzychb, 2016-11-30:
With commit https://github.com/espressif/esp-idf/commit/6e97936bace204d802cee8d34f6cbfed3a5dee0c documentation is reporting:

[image]

The same issue is flagged by Sphinx build:

```
... esp/esp-idf/docs/api/nvs_flash.rst:90: WARNING: doxygenfunction: Cannot find function "nvs_flash_init_custom" in doxygen xml output for project "esp32-idf" from directory: xml/
```

To prevent such issues from sneaking into documentation I propose to fail documentation build on Sphinx warnings (as already done for Doxygen build).

To fix documentation itself, instead of doing manual updates, I propose to automatically generate `apiname.rst` files or use `..  doxygenfile:: apiname.h` directive. The latter looks like the most attractive solution but depends on fixing of https://github.com/michaeljones/breathe/issues/287.

projectgus, 2016-12-22:
@krzychb , is this still an issue with the current Sphinx configuration?

krzychb, 2016-12-22:
@projectgus,

Specific warning in the title of this issue report is resolved. 

I assumed this waring to be an example of systematic problem of issues sneaking into documentation.
Basing on the list of warnings, I see for current esp-idf, this systematic problem is not resolved:

```
/home/krzysztof/esp/esp-idf/docs/api/bt.rst:4: WARNING: toctree contains reference to nonexisting document u'api/Bluetooth Classic'
/home/krzysztof/esp/esp-idf/docs/api/esp_gap_ble.rst:71: WARNING: doxygenstruct: Cannot find class "esp_ble_gap_cb_param_t" in doxygen xml output for project "esp32-idf" from directory: xml/
/home/krzysztof/esp/esp-idf/docs/api/esp_gatt_defs.rst:111: WARNING: doxygenstruct: Cannot find class "esp_gatt_rsp_t" in doxygen xml output for project "esp32-idf" from directory: xml/
/home/krzysztof/esp/esp-idf/docs/api/esp_gattc.rst:49: WARNING: doxygenstruct: Cannot find class "esp_ble_gattc_cb_param_t" in doxygen xml output for project "esp32-idf" from directory: xml/
/home/krzysztof/esp/esp-idf/docs/api/esp_gatts.rst:49: WARNING: doxygenstruct: Cannot find class "esp_ble_gatts_cb_param_t" in doxygen xml output for project "esp32-idf" from directory: xml/
/home/krzysztof/esp/esp-idf/docs/api/sigmadelta.rst:2: WARNING: Title underline too short.

Sigma-delta Modulation
========
/home/krzysztof/esp/esp-idf/docs/security/flash-encryption.rst:301: WARNING: Bullet list ends without a blank line; unexpected unindent.
```

igrr, 2016-12-22:
FYI, I haven't enabled warnings yet because I don't know yet how to include these 'unions of structs' defined in BLE headers into documentation properly.

closed 2017-01-16 by igrr