espressif/esp-idf #134: make menuconfig , file mconf.o Error1

Nourfdss, 2016-12-03:
Hello guys , please help me , im know using the chip im trying to make menuconfig in ubuntu but it shows errors ,the messages showed in the terminal :
make[1]: Entering directory `/home/nour/Desktop/iotbits_esp32/esp-idf/tools/kconfig'
cc  -DCURSES_LOC="<curses.h>" -DLOCALE   -c -o mconf.o mconf.c
mconf.c:21:20: fatal error: curses.h: No such file or directory
 #include <curses.h>
                    ^
compilation terminated.
make[1]: *** [mconf.o] Error 1
make[1]: Leaving directory `/home/nour/Desktop/iotbits_esp32/esp-idf/tools/kconfig'

thank u so much

andrei-ivanov, 2016-12-03:
Looks to me that you don't have all the prerequisites installed, in this case, the ncurses library.

nkolban, 2016-12-03:
See the following page:

http://esp-idf.readthedocs.io/en/latest/linux-setup.html

Notice `Step 0` instructions:

```
$ sudo apt-get install git wget make libncurses-dev flex bison gperf python python-serial
```

Were these executed?

Nourfdss, 2016-12-04:
Thanks guys now its working perfect and i can view the make menuconfig screen , but i have now new issue after  i typed  " make flash" i see this errors 
/home/noor/esp/esp-idf/make/component_common.mk:1: Deprecated feature: No longer necessary to include component_common.mk from /home/noor/esp/esp-idf/components/bt/component.mk
CC src/bootloader_flash.o
/home/noor/esp32/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/bin/as: error while loading shared libraries: libz.so.1: cannot open shared object file: No such file or directory
/home/noor/esp/esp-idf/make/component_wrapper.mk:176: recipe for target 'src/bootloader_flash.o' failed
make[2]: *** [src/bootloader_flash.o] Error 1
/home/noor/esp/esp-idf/make/project.mk:339: recipe for target 'bootloader_support-build' failed
make[1]: *** [bootloader_support-build] Error 2
/home/noor/esp/esp-idf/components/bootloader/Makefile.projbuild:28: recipe for target '/home/noor/esp/myapp/build/bootloader/bootloader.bin' failed
make: *** [/home/noor/esp/myapp/build/bootloader/bootloader.bin] Error 2
whats the problem ??

igrr, 2016-12-04:
This section of [Linux getting started guide](http://esp-idf.readthedocs.io/en/latest/linux-setup.html#step-1-download-binary-toolchain-for-the-esp32) may be what you are missing:

To use [the toolchain], you will need to update your PATH environment variable in ~/.bash_profile file. To make xtensa-esp32-elf available for all terminal sessions, add the following line to your ~/.bash_profile file:

export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin
Alternatively, you may create an alias for the above command. This way you can get the toolchain only when you need it. To do this, add different line to your ~/.bash_profile file:

alias get_esp32="export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin"
Then when you need the toolchain you can type get_esp32 on the command line and the toolchain will be added to your PATH.

Nourfdss, 2016-12-04:
@igrr hello , when i tried to type it shows me    bash: /home/noor/.bash_profile: Permission denied  , can i edited bashc instead of ~/.bash_profile ??

igrr, 2016-12-07:
Did you reload your profile script?
Before running `make`, verify that you can run `xtensa-esp32-elf-gcc` from command line.

Nourfdss, 2016-12-07:
How to verify that ? What should i do ? I run it in windows using the arduino ide , im using NodeMCU-32S Lua WiFi IOT Development Board , when i upload the file to esp32s i have to hold the boot button to load the code if not it will not load and give me error , but i wanna run in ubunto , is it the problem from the board ?? Thank u so much

igrr, 2016-12-07:
Type `xtensa-esp32-elf-gcc` on the command line. If you get an error that the command is not found, then you haven't configured your PATH environment variable correctly.

Nourfdss, 2016-12-07:
Yeah i wrote it , the result commnd not found , i tried again to edit the ~/.bash_profile   the result permission denied i tired to search online but i did not find to edit it this file ,, thanks for ur help

Nourfdss, 2016-12-07:
I edited the path in the file .bashrc but it seems doest work , i went to the home and press ctrl + H i found the .bash_profile and i typed the two path commnd still same problem ???

vitasimek, 2016-12-09:
Hi,
I had similar problem with gnome terminal and the debian.
I think this helped me: http://askubuntu.com/questions/132276/configure-gnome-terminal-to-start-bash-as-a-login-shell-doesnt-read-bashrc

 Try to add following lines to the .profile file located there in the home directory (it's hidden file):

export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin
export PATH=$PATH:$HOME/esp/crosstool-NG/builds/xtensa-esp32-elf/bin
export IDF_PATH=$HOME/esp/esp-idf

Check if the paths are correct for you.

Also there in the readme for linux is following instruction:

> Toolchain will be built in ~/esp/crosstool-NG/builds/xtensa-esp32-elf. Follow instructions given in the previous section to add the toolchain to your PATH.

Maybe it could be a bit misleading for someone, because you need to add /bin at the end to the path:

> $HOME/esp/crosstool-NG/builds/xtensa-esp32-elf/**bin**

vitasimek, 2016-12-09:
If you want use esptoo.py from command line, you cold fine usefull this:
export PATH=$PATH:$HOME/esp/esp-idf/components/esptool_py/esptool

Nourfdss, 2016-12-12:
yeah man now its working , but im doing flash i got an error makefile.projbuild recipe for target falsh faield , is it mean the terminal does not see the esp32 ???

projectgus, 2016-12-22:
I'm going to close this issue as it seems like most of the problems mentioned here have been solved. If there are errors during flashing then this is a different issue, searching esp32.com may help find solutions or please open a new issue with the error message.

closed 2016-12-22 by projectgus

rinelb, 2018-01-10:
what fix it for me was i didnt have the IDF_PATH in the C/C++ build -> Enviroment

dwitamghosh, 2018-01-15:
how to install ncurses librray for cygwin in windows??

Releece, 2019-03-27:
> how to install ncurses librray for cygwin in windows??

In Msys2 terminal execute: `pacman -S mingw32/mingw-w64-i686-ncurses`

I have the EXACT same error as the **issue opener**, however pacman returned ncurses was up to date, reinstalled anyway, no change in my error message. I completely reinstalled ESP-ADF which contains the ESP-IDF, also updated that ESP-IDF folder by emptying it and cloning anew through git, just to see if it would fix it, but nope. Updated Msys2 through `pacman -Syu`, no difference. What problem can it be that no make command works anywhere in a completely updated and fresh ennvironment, trying examples right out of the package? There could be no old header references such as in https://github.com/espressif/esp-idf/issues/712 (tried everything in that thread too)

Git submodules -- init --recursive always fails with error