espressif/esp-idf #135: VFS: Unable to map the whole file system tree using "/"

nkolban, 2016-12-03:
I am finding that I am unable to map the whole file system tree as seen by the posix APIs using esp_vfs_register().  It appears that if I try and map "/" that returns an error code.  It is as though the esp_vfs_register() requires that my mount point be subordinate to root ("/").  I am thinking that this may be too restrictive.  What if I want to use VFS with just a single type of file system (for examples SPIFFS) then I want all paths to map into the SPIFFS file system not just those from a specific prefix.

For example, I want to be able to access "/dir1/file1.txt" and "/dir2/file2.txt" through SPIFFS with both files on the same SPIFFS file system.

negativekelvin, 2016-12-03:
You have no options to set base path in your code?

jmattsson, 2016-12-05:
I imagine this is related to the fact the VFS does not support nested mount points, and therefore would break e.g. the `/dev/uart` registration.

igrr, 2016-12-07:
I can implement nested mount points if there are some use cases for that.

rojer, 2016-12-13:
i just filed #158 which is related (feel free to dedup) - need to be able to take over the entire vfs.

igrr, 2016-12-13:
Adding a request from @rojer here: we should support paths which don't start with `/`.

rojer, 2016-12-13:
fwiw, above is my stab at it.

closed 2017-06-23 by igrr

rojer, 2017-06-23:
@igrr almost there, but relative paths like `foo` or `conf.json` are not passed to the default VFS.
https://github.com/espressif/esp-idf/pull/726 makes that happen.

igrr, 2017-06-26:
Good catch, thanks @rojer.