espressif/esp-idf #136: rtc_pads_slpie() and rtc_pads_funie() need to support RTC_GPIO0 - RTC_GPIO6

comcat, 2016-12-03:
rtc_pads_slpie() and rtc_pads_funie() in current version of librtc.a is not support to setup the SLP_IE/FUN_IE of RTC_GPIO0 ~ RTC_GPIO6:

   0x4008da1c <rtc_pads_slpie>:   entry   a1, 32
   0x4008da1f <rtc_pads_slpie+3>: bbci    a2, 6, 0x4008da40 <rtc_pads_slpie+36>
   0x4008da22 <rtc_pads_slpie+6>: l32r    a8, 0x4008bfa4
   0x4008da25 <rtc_pads_slpie+9>: l32r    a9, 0x4008da18
   ......

There is bits of SLP_IE and FUN_IE in the registers of RTC_GPIO0 ~ RTC_GPIO6 (RTCIO_SENSOR_PADS_REG and RTCIO_ADC_PAD_REG)

This one cause the rtc_pad_ext_wakeup() can not work for RTC_GPIO0 - RTC_GPIO6, need a fixup like: REG_SET_BIT(RTC_IO_ADC_PAD_REG, RTC_IO_ADC2_SLP_IE);

comcat, 2016-12-03:
More info please refer to: http://wiki.jackslab.org/ESP32_RTC_External_Wakeup

igrr, 2016-12-04:
I'll keep this issue open for now, but please take into account the fact that said functions are not part of ESP-IDF API, and external wakeup from deep sleep is not a supported feature yet. We will remove parts of this library on the next release.

comcat, 2016-12-05:
Unable to make excessive demands to the guys who write the librtc.a ...there are 17 input parameters of rtc_sleep_prep()... :(
close this issue to wait a normal release

closed 2016-12-05 by comcat