espressif/esp-idf #139: RSA hardware assist fails with 4096 bit RSA certificate - "too large for hardware unit"

ggsubs, 2016-12-04:
With RSA Hardware assist the 04_https_request example fails to connect to a host with 4096 bit RSA cert:

```
I (5346) example: Connecting to api.gigafive.com:443...
I (6076) example: Connected.
I (6076) example: Performing the SSL/TLS handshake...
E (6166) bignum: ERROR: 4608 bit result 4127 bits * 32 bits too large for hardware unit

E (6166) example: mbedtls_ssl_handshake returned -0x428e
E (6166) example: Last error was: -0x428e - RSA - The public key operation failed : BIGNUM - The input arguments are not acceptable
```

ggsubs, 2016-12-04:
As a workaround, hardware MPI can be turned off:

```
#
# mbedTLS
#
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=16384
CONFIG_MBEDTLS_DEBUG=y
CONFIG_MBEDTLS_HARDWARE_AES=y
# CONFIG_MBEDTLS_HARDWARE_MPI is not set
CONFIG_MBEDTLS_HARDWARE_SHA=y
```

projectgus, 2016-12-04:
Thanks for reporting this. I can confirm this issue, we'll update you when we have a fix or a workaround.

You can also find the option to disable hardware big number operations in menuconfig under "Component Config -> mbedTLS -> Disable hardware MPI (bignum) acceleration"

closed 2017-03-06 by igrr

holzachr, 2017-03-07:
Thanks for fixing this issue - I've seen you also added some test code, which will probably work fine.
Anyway, in combination with mbedTLS, MPI acceleration enabled and 4096 bit keys, code hangs infinitely at

`I (6042) example: Performing the SSL/TLS handshake...`

Works fine for 2048 bit keys though.

Anyone out there to confirm the issue?