espressif/esp-idf #140: Eighty seven warnings in esp-idf release 1.0

krzychb, 2016-12-04:
Hi,

Congratulations on releasing esp-idf 1.0 including extensive documentation!

After clean build of of `01_hello_world` example I see almost 90 warnings.
Most of them coming from component/bt.

Do you see the same?

This number of warnings is distinctive so one can miss warnings flagged for own code.

[image]

jrobeson, 2016-12-04:
maybe such checks should be part of the CI system?

igrr, 2016-12-04:
This was supposed to be part of CI check, but was cleverly worked around in component makefile.

https://github.com/espressif/esp-idf/blob/master/make/build_examples.sh#L24

We are tracking this issue internally and will have a fix soon.

projectgus, 2016-12-22:
This should be fixed, including a CI check for warnings to avoid regressions.

Please open if you see other spurious warnings appearing.

closed 2016-12-22 by projectgus