espressif/esp-idf #141: Crash when using vTaskSuspend

vanthome, 2016-12-04:
Using vTaskSuspend in a task `vTaskSuspend(NULL);` leads to:

```
V (20268) event: exit default callback
Untested FreeRTOS function vTaskSuspend
/esptest/esp-idf/components/freertos/./tasks.c:1695 (vTaskSuspend)- assert failed!
Guru Meditation Error: Core   0 panic'ed (Unhandled debug exception)
Register dump:
PC      :  40008155  PS      :  00060936  A0      :  80007d16  A1      :  3ffcb1b0  
A2      :  0000005f  A3      :  0000005f  A4      :  00000010  A5      :  ffffffff  
A6      :  00000000  A7      :  fffffff3  A8      :  00000000  A9      :  3ffcb120  
A10     :  00000000  A11     :  3f403428  A12     :  3ffcb0ef  A13     :  00000035  
A14     :  00000000  A15     :  3ffcb0f5  SAR     :  00000004  EXCCAUSE:  00000001  
EXCVADDR:  00000000  LBEG    :  00000000  LEND    :  00000000  LCOUNT  :  00000000  
Rebooting...
ets Jun  8 2016 00:22:57
```

negativekelvin, 2016-12-05:
https://github.com/espressif/esp-idf/blob/65f57e5da7287576d98f657ea45163411bc3a564/components/freertos/Kconfig#L47

igrr, 2016-12-07:
Closing as duplicate of #107.

closed 2016-12-07 by igrr