espressif/esp-idf #143: Cant configure gpio33 to input or output

lollizze, 2016-12-05:
Hi
Since the release v1.0 i can't configure gpio33 as an output nor an input. 

Just to be sure i didn't introduce something i also tried it with the 02_blink example with no luck
I tried some other random pins
led blinks for pins: 21, 23, 25, 27 
nothing for pins: 32, 35

igrr, 2016-12-05:
AFAIK gpio32 and 33 are configured as XTAL pins by default. To re-connect them to the digital subsystem, you need to clear the RTCIO_XTAL_X32N_MUX_SEL and RTCIO_XTAL_X32P_MUX_SEL bits.

comcat, 2016-12-06:
GPIO32 and 33 are routed to RTC domain by default. The default value of the control register RTC_IO_XTAL_32K_PAD_REG is 0x84160010, X32P_MUX_SELand X32N_MUX_SEL are set to 1 (route to RTC, the description of these bit in the header file are wrong)

Need a setup something like this:

/* gpio32 route to digital io_mux */
REG_CLR_BIT(RTC_IO_XTAL_32K_PAD_REG, RTC_IO_X32P_MUX_SEL);

/* gpio33 route to digital io_mux */
REG_CLR_BIT(RTC_IO_XTAL_32K_PAD_REG, RTC_IO_X32N_MUX_SEL);

Testing pass

lollizze, 2016-12-06:
hm, i thought it had something to do with the 32k_xn, but couldn't find what, thanks!

Would have been great if say gpio_config would clear those bits as well, or maybe get some errors

lollizze, 2016-12-06:
Still probleme with gpio33....
It worked in the example, but not in my code.....
The difference is I use gpio_config to configure the gpio, but when i do the device panics, and restart.

So added that in the blink example, and i get the following error
Guru Meditation Error of type LoadProhibited occurred on core   0. Exception was unhandled.
Register dump:
PC      :  400d8d06  PS      :  00060730  A0      :  800d8747  A1      :  3ffb7990
A2      :  3ffb79f0  A3      :  00000021  A4      :  3ff49020  A5      :  00000000 
A6      :  00000002  A7      :  00000020  A8      :  3f403e58  A9      :  00000000
A10     :  00000000  A11     :  00000001  A12     :  3ffb79b0  A13     :  0000000c
A14     :  3ffb57a4  A15     :  3ffb576c  SAR     :  0000001f  EXCCAUSE:  0000001c
EXCVADDR:  00000000

here is what i do before the panic:
REG_CLR_BIT(RTC_IO_XTAL_32K_PAD_REG, RTC_IO_X32P_MUX_SEL);
REG_CLR_BIT(RTC_IO_XTAL_32K_PAD_REG, RTC_IO_X32N_MUX_SEL);
/* Set the GPIO as a push/pull output */
config.pin_bit_mask = ((uint64_t)1) << BLINK_GPIO;
config.intr_type = GPIO_INTR_DISABLE;
config.mode = GPIO_MODE_OUTPUT;
config.pull_up_en = GPIO_PULLUP_DISABLE;
config.pull_down_en = GPIO_PULLDOWN_DISABLE;
status = gpio_config(&config);

rdpowers, 2016-12-07:
@lollizze, also note that GPIO34 & GPIO35 (aka VDET_1, VDET_2) are not possible to use as general purpose outputs (TRN v1.1 Page 39), so Blink won't ever work on them. Same goes for SENSOR_* pins.

In other words, GPIO34-39 should more properly be called GPI because they can only be used as inputs.

igrr, 2016-12-08:
This is now fixed in master (if you use `gpio_pullup_en`/`gpio_pullup_dis`/... APIs).

closed 2016-12-08 by igrr