espressif/esp-idf #146: Internal Temp Sensor API

forthlightning, 2016-12-06:
I am wondering when there will be support for the internal temperature sensor? 

I'm hoping to do some internal diagnostics on my project, please let me know!

vitasimek, 2016-12-06:
Hi,
I am also interested in this. I have read all the doc available, I hope, and I think it is (?) the part of the ADC stuff, that has been already merged there in the 1.0 release. Is it true? Well, if so, I can try to help to hack it to get it working somehow.

forthlightning, 2016-12-06:
Yeah i looked through the code and it looks like they have the registers all mapped out [here](https://github.com/espressif/esp-idf/blob/4480ab6c8ce70778df4948e587e712540de237eb/components/esp32/include/soc/saradc_reg.h)

vitasimek, 2016-12-06:
Well, I don't know exactly if this matters, but I've found this video about ESP32 beta and ADC1 by rudi: https://www.youtube.com/watch?v=hVLhqiNS3vU

It looks like he may to know how it works. Please correct me if I am wrong.

vitasimek, 2016-12-06:
And the investigation continues to here: http://esp32.com/viewtopic.php?f=2&t=494&p=2121#p2121

vitasimek, 2016-12-07:
There has to be  some mechanism for starting ADC conversion, waiting and then reading the result from some register. I have found just this, that maybe could be some clue: SARADC_SAR_START_FORCE_REG there in the ulp.c file.

projectgus, 2016-12-07:
The SAR ADC code is in code review right now, so there will be answers very soon! Sorry for the delay on this one.

vitasimek, 2016-12-07:
I just got working the rudi's example with ADC. 

@projectgus: I know you are working really really hard and it deserves big respect. Thank you for your effort.

Is there some register address (like ADC1_GPIO34) that is associated with the onboard temperature sensor?

comcat, 2016-12-07:
on chip sensor, try this: http://wiki.jackslab.org/ESP32_Onchip_Sensor

ADC: http://wiki.jackslab.org/ESP32_ADC

vitasimek, 2016-12-07:
Hi Jack,
your example just works! :)
The functions are already there in the libs, the only thing is to define them as they would been defined there in the header file.

Thank you.

yschaeff, 2016-12-07:
Also, while you are at it fix the spelling. temprature_sens_read() should read temperature_sens_read(). I was wondering why my greps didn't yield any result.

forthlightning, 2017-03-21:
bump :D

HubbyGitter, 2017-06-27:
I recommend against using `temprature_sens_read()`. Use the code from [`test_tsens.c`](https://github.com/espressif/esp-idf/blob/master/components/esp32/test/test_tsens.c) (at least as a basis) instead.

Regarding `temprature_sens_read()`:

- It is an undocumented an unofficial function.
- Its initialisation sequence is not clean.
- It yields readings which deviate from that of the implementation in `test_tsens.c`, presumably due to a very fast conversion clock. However, you won't get a faster reading than with the code in `test_tsens.c`.
- It uses busy waiting without you even knowing.

Please refer to [this discussion](https://github.com/espressif/esp-idf/issues/732) which describes more background and a possible modification to the code in `test_tsens.c`.

HubbyGitter, 2017-06-27:
As requested in the discussion mentioned in my previous comment, I'm adding the following information here.

The following two screenshots show a comparison of TSENS readings via `temprature_sens_read()` ("n/a 2") and via `test_tsens.c` code ("it"), as well as a precision external sensor (BME280). All data points are taken "simultaneously" and posted as a set. And yes, the fan which I used for the cooling disturbed the WiFi so not all data points were captured here towards the right side.

I think this shows that readings via `temprature_sens_read()` have significantly more jitter, generally tend to "swing" / overreact, and have problems reaching higher values -- also supported by past observations. (Screenshot 1 starts directly after a flashing procedure which led to increased chip temperature.)

[image]

[image]

mvadu, 2017-12-23:
@me-no-dev  Its been 6 months since last activity in this topic. Is there any standard API (documented, usable in Arduino) that espressif recommends to make use of the temp sensor?

projectgus, 2019-05-15:
Unfortunately due to performance issues the temp sensor feature is no longer supported.

closed 2019-05-15 by projectgus