espressif/esp-idf #147: 09_openssl_client Exceptions

ksitko, 2016-12-06:
I can't get example 09 to work without getting a exception fault:

```
.[0;32mI (7337) Openssl_demo: OpenSSL demo thread start OK.[0m
.[0;32mI (7337) Openssl_demo: get target IP address.[0m
.[0;32mI (7417) Openssl_demo: OK.[0m
.[0;32mI (7417) Openssl_demo: 103.235.46.39.[0m
.[0;32mI (7417) Openssl_demo: create SSL context .......[0m
.[0;32mI (7427) Openssl_demo: OK.[0m
.[0;32mI (7427) Openssl_demo: create socket .......[0m
.[0;32mI (7427) Openssl_demo: OK.[0m
.[0;32mI (7437) Openssl_demo: bind socket .......[0m
.[0;32mI (7437) Openssl_demo: OK.[0m
.[0;32mI (7437) Openssl_demo: socket connect to remote www.baidu.com .......[0m
Guru Meditation Error of type InstrFetchProhibited occurred on core   0. Exception was unhandled.
Register dump:
PC      :  00000001  PS      :  00060d30  A0      :  800f0646  A1      :  3ffc4e50  
A2      :  00000001  A3      :  3ffca374  A4      :  00000001  A5      :  00000008  
A6      :  00000000  A7      :  3ffbdbf8  A8      :  800eed30  A9      :  00000002  
A10     :  3ffca374  A11     :  00000001  A12     :  3ffbf6b0  A13     :  00000000  
A14     :  3ffc2b2c  A15     :  3ffbf624  SAR     :  00000013  EXCCAUSE:  00000014  
EXCVADDR:  00000000  LBEG    :  4000c2e0  LEND    :  4000c2f6  LCOUNT  :  ffffffff  
Rebooting...
ets Jun  8 2016 00:22:57
```

ether42, 2016-12-07:
Panic is similar to #133 (PC, InstrFetchProhibited, ...), you may want to take a look at it.

igrr, 2016-12-08:
@ksitko This may have been fixed in master, could you please check if you are still able to reproduce?

projectgus, 2016-12-22:
I'm going to close this for now, but please reopen if you try with latest master and the crash reoccurs.

closed 2016-12-22 by projectgus