espressif/esp-idf #150: missing dependency for running gdb on Arch linux

dumarjo, 2016-12-07:
Hi ,

I try to run gdb on my arch linux and gdb need some dependency that are not found on arch linux. I followed the readme to install the linux setup + eclipse front end. Up to now I can compile and flash my bord from linux. 

I try to get ARM-USB-OCD working on my ESP32 thing board (from sparkfun). I have wired it already and openOCD seem to be happy with it. 

Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Core was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32.cpu0: Target halted, pc=0x40080AC8
esp32.cpu0: target state: halted

When I try to start gdb, I got those error messages. 

could not found shared library libncurses.so.5 and libtinfo.so.5

ldd xtensa-esp32-elf/bin/xtensa-e
	linux-gate.so.1 (0xb77cf000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0xb77a2000)
	libncurses.so.5 => not found
	libtinfo.so.5 => not found
	libm.so.6 => /usr/lib/libm.so.6 (0xb774c000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0xb772f000)
	libpython2.7.so.1.0 => /usr/lib/libpython2.7.so.1.0 (0xb754d000)
	libc.so.6 => /usr/lib/libc.so.6 (0xb7394000)
	/lib/ld-linux.so.2 (0xb77d1000)
	libutil.so.1 => /usr/lib/libutil.so.1 (0xb7390000)

On Arch I have 

/usr/lib/libncurses++.so
/usr/lib/libncurses++w.so
/usr/lib/libncurses++w.so.6
/usr/lib/libncurses++w.so.6.0
/usr/lib/libncurses.so
/usr/lib/libncursesw.so
/usr/lib/libncursesw.so.6
/usr/lib/libncursesw.so.6.0

But I don't have any libtinfo.so

What should I do now ?

any idea ?

projectgus, 2016-12-07:
For ncurses5 (including libtinfo) on Arch, there is a package from AUR that is needed:
https://aur.archlinux.org/packages/ncurses5-compat-libs/

Or possibly, as you are using the 32-bit toolchain:
https://aur.archlinux.org/packages/lib32-ncurses5-compat-libs/

I will update the prerequisites guide to mention this.

dumarjo, 2016-12-07:
Hi,
I will have a look at it. 

Regards
Jonathan

dumarjo, 2016-12-07:
lib32-ncurses5-compat-libs 6.0-4  (2016-07-07 23:34)
( Unsupported package: Potentially dangerous ! )
==> lib32-ncurses5-compat-libs dependencies:
 - lib32-glibc (building from AUR)
 - lib32-ncurses (building from AUR)
 - gcc-multilib (building from AUR) [makedepend]

==> Continue building lib32-ncurses5-compat-libs ? [Y/n]
==> ----------------------------------------------------
==> 
==> Building and installing package
==> Install or build missing dependencies for lib32-ncurses5-compat-libs:

error: target not found: gcc-multilib
error: target not found: lib32-ncurses
error: target not found: lib32-glibc
==> Restart building lib32-ncurses5-compat-libs ? [y/N]
==> ---------------------------------------------------
==> 

regards
Jonathan

dumarjo, 2016-12-07:
Hi,

Building it from your crossNG fork seem to work. No problem starting gdb.

Regards

Jonathan

closed 2016-12-30 by projectgus