espressif/esp-idf #153: Loading application from openOCD

dumarjo, 2016-12-09:
Hi,
Is it possible to upload the firmware through GDB ? When I try it, I got this error from openOCD

Error: esp32.cpu0: xtensa_write_memory (line 1024): DSR (8020CC13) indicates DIR instruction generated an exception! Warn : esp32.cpu0: Failed writing 4096 bytes at address 0x3F400010

Regards

Jonathan

Spritetm, 2016-12-09:
Not yet; OpenOCD is not aware that the application lives in flash and will try to write to the flash as if it's RAM. Remedying this is on our ToDo-list, but we haven't gotten around to this.

dumarjo, 2016-12-09:
Hi,

Thanx for the Info. 

Jonathan

closed 2016-12-09 by dumarjo

dantonets, 2017-05-17:
Hi,
I got this problem as well. The extensa-gdb works OK when start from the command line. But it fails with this message when I try start it from eclipse (Neon.3) - it try write to the drom0_0_seg, and, obviously fails - it read-only memory.  So, can somebody point me to the eclipse configuration for avoid this write?
Thanks.

dantonets, 2017-05-18:
Never mind, I find a fix for that ..Thanks.