espressif/esp-idf #158: Default VFS

rojer, 2016-12-13:
I need to be able to take over the entire filesystem, i.e. i want my code to be responsible for all paths, not just some `/subset`.
I am fine with the VFS interface as such, it looks good, but this calls for some sort of a "default vfs", or "vfs of last resort" perhaps. E.g. if i register filesystem with an empty prefix, then in absence of a more specific match to should fall through to that implementation.

igrr, 2016-12-13:
Would writing your own handlers into syscall table during application startup work for you? I.e. modify `syscall_table_ptr_pro` and `syscall_table_ptr_app` to point to your functions.

This approach isn't perfect because VFS stuff will still be linked into the application. We can maybe introduce some menuconfig option to disable VFS completely.

rojer, 2016-12-13:
that'd be fine too, as long as the interface remains stable.
but i actually like the facility provided by the VFS library - namely, translating FDs and providing context pointers. i'd have to reimplement that. a "catch all" vfs seems preferable to me.

igrr, 2016-12-13:
Okay, i understand. In that case implementing #135 in ESP-IDF would be a better option. I'll work on it once I finish all the current stuff with RTC/ULP (in a week or so).

rojer, 2016-12-13:
ok, thanks! for now i'll implement my own kludge to do it.
one last thing: in my world paths don't start with a `/` (legacy), so it really has to be everything.

(closing in favor of #135)

closed 2016-12-13 by rojer