espressif/esp-idf #159: EAP problem, always try to assoc with anonymous@espressif.com

ivanO86, 2016-12-13:
Hi all,

I am having problems when try to connect via WPA-enterprise. In the first stage of the EAP TTLS phase, the device always tries to assoc with anonymous@espressif.com user instead of with the user I set before. I don't know if is a problem of my config or is an standar behaviour. Here there is the code to set the user/pass config for WPA-entreprise:

esp_wifi_sta_wpa2_ent_enable();
/*sanity clean*/
esp_wifi_sta_wpa2_ent_clear_username();
esp_wifi_sta_wpa2_ent_clear_password();
esp_wifi_sta_wpa2_ent_clear_ca_cert();
/*set credentials*/
esp_wifi_sta_wpa2_ent_set_username(user, strlen(user));
esp_wifi_sta_wpa2_ent_set_password(pass, strlen(pass));			esp_wifi_sta_wpa2_ent_set_ca_cert(ca, strlen(ca));
/*connect*/
...

Did I forget any step?

ivanO86, 2016-12-13:
Sorry, there have been an misunderstood. I mind that I want to avoid the use of the anonymous behaviour and use my own anonymous user.

igrr, 2016-12-13:
Thanks for the report, an API to set user id should be added soon.

joostd, 2017-01-15:
Update to get the latest version of libwpa2 and use
```
esp_err_t esp_wifi_sta_wpa2_ent_set_identity(unsigned char *identity, int len);
```
to set the EAP outer identity

closed 2017-01-16 by igrr