espressif/esp-idf #160: Build error when using inet_ntop macro

jolivepetrus, 2016-12-14:
In this code:

	struct sockaddr_in *sin = (struct sockaddr_in *)sa;
	inet_ntop(sin->sin_family, &sin->sin_addr, addr_string, ADDRLEN);
	sprintf(&addr_string[strlen(addr_string)], ":%d", ntohs(sin->sin_port));

We get this compilation warning:

/Users/jaumeolivepetrus/esp-idf/components/lwip/include/lwip/lwip/sockets.h:603:15: error: 'AF_INET6' undeclared (first use in this function)
     (((af) == AF_INET6) ? ip6addr_ntoa_r((const ip6_addr_t*)(src),(dst),(size)) \
               ^
/Users/jaumeolivepetrus/Lua-RTOS-ESP32/components/mqtt/./Socket.c:842:2: note: in expansion of macro 'inet_ntop'
  inet_ntop(sin->sin_family, &sin->sin_addr, addr_string, ADDRLEN);
  ^
/Users/jaumeolivepetrus/esp-idf/components/lwip/include/lwip/lwip/sockets.h:603:15: note: each undeclared identifier is reported only once for each function it appears in
     (((af) == AF_INET6) ? ip6addr_ntoa_r((const ip6_addr_t*)(src),(dst),(size)) \

projectgus, 2016-12-21:
Hi @jolivepetrus ,

Sorry for the delay response. I'm not able to reproduce this, have attached a very minimal "main.c" file from the idf-template project which compiles correctly (the code attached will not even remotely work if run, but that's a secondary issue!)

Looking at the sockets.h header it looks like AF_INET6 should be defined in all cases, to either "10" or to AF_UNSPEC (0):
https://github.com/espressif/esp-idf/blob/master/components/lwip/include/lwip/lwip/sockets.h#L209

Is it possible something else in the compilation unit is `#undef AF_INET6` for some reason? Or is there a bug in the way sockets.h is included?

projectgus, 2016-12-22:
Oops, can't attach source files. Here's a gist: https://gist.github.com/projectgus/57d206a04ea4d59150d70e36b446dc83

projectgus, 2016-12-30:
Closing for now, but @jolivepetrus please reopen if this is stil an issue for you.

closed 2016-12-30 by projectgus

jolivepetrus, 2016-12-30:
Sorry for the delay in my answer.

We used our implementation of MQTT for PIC32 which had an #undef AF_INET6.
 
Sorry for the inconvenience.