espressif/esp-idf #161: Wifi GOT_IP event is delivered to system handler, not wifi-specific

rojer, 2016-12-14:
I like ability to set a separate event handler when doing wifi_init - it helps isolate functionality and keep wifi stuff in its own module.
This works as expected for `SYSTEM_EVENT_STA_START` and `SYSTEM_EVENT_STA_CONNECTED` - they are delivered to the wifi callback. However `SYSTEM_EVENT_STA_GOT_IP` is delivered to the system event handler instead.

nkolban, 2016-12-15:
Howdy ... in my projects, I have created an alternate event handler and found that all events were delivered to it instead of the original event handler.  I had no issues.  Would you be able to further expand on your issue with more of a description of the logical design you are using and what you experienced.  If I can be of assistance in the IRC `#ESP32` chat room, feel free to ping me as user `kolban`.

igrr, 2016-12-16:
Right... I think we can provide the callback as an argument when initializing tcpip_adapter.
We would have to change `tcpip_adapter_init` signature, but we have to break a few other APIs in 2.0, so I guess that's not a huge problem.

jack0c, 2018-03-21:
@igrr  Shall we close this issue?

rojer, 2018-03-21:
yeah, this is no longer an issue for me.

closed 2018-03-21 by rojer