espressif/esp-idf #162: AP: Add a way to control which DNS server is offered to clients

rojer, 2016-12-14:
At the moment from reading the code i see that device's address is always offered as the DNS server.
There should be a way to control whether server is offered at all and if so, which IP.

liuzfesp, 2017-06-07:
HI @rojer, a new menuconfig is added to configure whether the default dns server will be configured or not, if enabled, the dns server address got from DHCP ACK's option "Host Domain Name" will be set as default DNS server, otherwise, the address in DHCP Ack's option will be ignore and the user need to configure the DNS server. I think the code will be ready in next release, maybe release2.1

rojer, 2017-06-07:
this issue is about which server ESP AP offers to clients, i.e. the behavior of its DHCP server, while you seem to be talking about ESP's STA DHCP client behavior.

the original reason for this issue is that it is not always that when i connect to ESP AP i also want my traffic to go trough it (router option) or for it to be used a s a DNS server (DNS option). in fact, i'd argue that in most cases AP is used to provision the device, so it should default to no routing and no DNS, so that user retains their original internet connection.

rojer, 2017-06-07:
in any case, it should be possible to turn off DNS and router options in the server responses. it's already possible with router, DNS still remains.

liuzfesp, 2017-06-07:
Thanks @rojer for clarification, then I'm going to add options to DNS server option for AP side.

rojer, 2017-06-07:
good, thanks!

FayeY, 2017-07-05:
Is this problem solved in the latest ESP-IDF?

rojer, 2017-07-05:
as far as i can tell, no. @liuzfesp ?

closed 2017-10-23 by igrr