espressif/esp-idf #164: ESP32 ADC precision

nassks, 2016-12-14:
Hi,

I'm testing the ESP32's ADC on the SparkFun ESP32 Thing.
Currently my measure seems very noisy, it constantly varies on the 5 first bits, and the zero value is around 90mV.
Is there a way to achieve better accuracy?

My test code : 
```
adc1_config_width(ADC_WIDTH_12Bit);
adc1_config_channel_atten(ADC1_CHANNEL_7, ADC_ATTEN_0db);

while (true) {
        vTaskDelay(300 / portTICK_PERIOD_MS);
        printf("ADC value : %x\n", adc1_get_voltage(ADC1_CHANNEL_7));
}
```

Thank you in advance for your help.

projectgus, 2016-12-15:
The ADC has a non-linear response, and we are working on producing characterisation data which will be added to esp-idf to produce a linear output.

When you say "noisy", are you getting noisy readings on a constant voltage or is this noise caused by the non-linearity?

There's some more in this related Arduino bug: https://github.com/espressif/arduino-esp32/issues/92

CarlosGS, 2016-12-16:
> There's some more in this related Arduino bug: espressif/arduino-esp32#92

Hi, a bit off-topic but I didn't want to start a new thread:

How come the adc2 function headers are not defined in adc.h? Won't adc2
work in the same way as adc1?

I see the adc2 functionality has already been implemented in the arduino
kernel, and it looks to me it works exactly as the adc1.

Also, thanks for the great work. I'm loving the ESP32 so far :-)

projectgus, 2016-12-19:
@CarlosGS : for issues, it is much better to start a new one than to post off-topic in an existing one.

For questions like this, the http://esp32.com forum may be an even better choice.

michapr, 2016-12-27:
@projectgus : I think nassks means the "noise" within same values - additional I have seen that there is a "noise" between the inputs.
put 1 Volt (with capacititor) on inputs CH4 ..CH7 I get such results:
(using 0bd and 11db for the values)
ADC1 CH4 value: 3719 @ 0db, 1056 @ 11db
ADC1 CH5 value: 3889 @ 0db, 1052 @ 11db
ADC1 CH6 value: 3883 @ 0db, 1052 @ 11db
ADC1 CH7 value: 3891 @ 0db, 1052 @ 11db
-------
ADC1 CH4 value: 3867 @ 0db, 1056 @ 11db
ADC1 CH5 value: 3885 @ 0db, 1053 @ 11db
ADC1 CH6 value: 3887 @ 0db, 1040 @ 11db
ADC1 CH7 value: 3886 @ 0db, 1057 @ 11db
-------
ADC1 CH4 value: 3894 @ 0db, 1056 @ 11db
ADC1 CH5 value: 3855 @ 0db, 1061 @ 11db
ADC1 CH6 value: 3887 @ 0db, 1049 @ 11db
ADC1 CH7 value: 3875 @ 0db, 1053 @ 11db
-------

Difference is up to 5% - maybe it is normal?

But for 0V - I get a value "0" ...

Michael

michapr, 2016-12-27:
Do not know why the crazy style came up ;)

Michael

marcmerlin, 2017-03-28:
So, in case anyone finds this ADC bug, I've found another ADC issue apparently where it will return data between 0-50% input, and cap out at 4096 from 50 to 100% input, but only after I try to use RMT.
Details here: https://github.com/espressif/esp-idf/issues/462

marcmerlin, 2017-04-04:
So, the lastest arduino-esp32 fixes the issue I just reported when RMT is turned on, analogread values would get damaged.

FayeY, 2017-05-04:
Hi nassks, is this problem solved? Can we close this issue?

igrr, 2017-05-04:
@FayeY not yet, @Spritetm is working on it.

dmody, 2017-05-06:
Just leaving my two cents on this...
I was hoping to use an inexpensive Voltage Reference IC (LM4040C30ILP) to help increase the accuracy of measured values.  I too found the ADC to be non linear and have included my data below for anyone interested.
voltage	ADC
2	2222
3	3730
1	1033
1.5	1629
0	0
0.5	422
2.5	2840
2.8	3300
3.1	3980
3.007	3750

[image]

dmody, 2017-05-07:
took another crack at this, and assuming the ADC readings are related to the 3.3V regulator voltage (mine is at 3.275 V on a developer board).  Here's what I found.
[image]

negativekelvin, 2017-05-07:
@dmody interesting, have you tried to fit curves to the upper range and lower range separately?

imotion-software, 2017-05-11:
In the future, are you planning to include linearization of adc1_get_voltage or it will be in charge to the user to correct non linear behavior?

Just to know, because if we implement linearization on user app and then adc1_get_voltage will change behavior we'll have to remove all those parts.

dmody, 2017-05-11:
I did not consider doing that.  Sounds like a decent idea.  I wonder which
would be faster to execute.  Sounds like a fun experiment to see.

I think the bigger problem I'm having is the noise.   I haven't put my
scope on the unit to see if the noise is real or not, but my approach to
filtering was to do a for loop acquiring 100 samples, and to do a moving
average of 10 samples.
Ie
For I= 1 to 100
  Daq = (analog input value + daq*9)/10

Would love to hear other people's thoughts on this.

markterrill, 2017-06-18:
really interested in this

al3x-huang, 2017-06-22:
Are we going to fix this soon? I feel like many applications depend on this feature. Is there a workaround at least?

Spritetm, 2017-06-23:
Yes, we are. It took us some time to build an automated testing rig to accurately and reliably grab the characteristics of a representative sample of ESP32s; that has rendered the first curves this week and we're working on converting these into some linearization code.

al3x-huang, 2017-06-30:
@Spritetm any updates on this? Perhaps a timeline to when this might be fixed? Thanks!

MaxSNi, 2017-07-28:
@Spritetm @igrr I would love to hear an update on this - or if there is an effective workaround?

MaxSNi, 2017-07-28:
@dmody I would love to see your function. I made a quick function from my own excel trend equation and while it does work better, its pretty crude...

dmody, 2017-07-29:
@MaxSNi 

//*************************************** Analog to Digital Conversion Function *********************************
float A2D(float SenVal)
{
  if(SenVal<1) return(0);
  else if (SenVal>4094) return(99);
  else  return(-0.000000000023926 * pow(SenVal,3) + 0.000000094746 *pow(SenVal,2) + 0.00074539*SenVal + 0.14925);

}

imotion-software, 2017-08-03:
Is there any update on this issue?
It sounds me really crazy that on an IoT chip there's no way to have a good ADC reading.

markterrill, 2017-08-03:
Probably time to look at other options.
i realised a few months ago its simply not going to be ready anytime this calendar year for production use

jan-bozelie, 2017-08-21:
we have a simmulair problem , i can live with a correction factor  but we discoverd in testing 10 esp32 samples , the difference between samples can be >10 to 20%  , making a calibration nessesery  for each esp , making it unusable for serios work

imotion-software, 2017-08-22:
Hi @jan-bozelie. I'm aware there should be huge differences betweens samples due to the internal reference voltage regulator, but at the moment ADCs suffer of two distinct issues:

- Non linearities
- Different end of scale due to internal voltage reference

I think we should split ADC in two parts. Once corrected the non linearities, the EOS issue could be faced easily calibrating with an external voltage reference.

This problem has been totally ignored by Espressif, and I think ADCs are really important for an IoT application.

@igrr ... can we have an official response from Espressif about the fact ADC issue can or cannot be solved? At the momento we have a peripheral which is absolutely useless.

igrr, 2017-08-22:
The official response has been already given by @Spritetm above (well, that is as official as you will get). We are working on both parts of the issue (nonlinearities and different v_ref):
- For non linearity, the code and linearization tables are being reviewed at the moment and as mentioned somewhere (i think on the forum) these will be part of 3.0 release.
- For v_ref variation, there is a provision in the linearization code to take v_ref value for the current chip into account. Obviously you need to determine this value somehow somehow. There are three ways of solving it:
  - Large scale: new chips produced starting from (some date)(1) will have their V_REF burned into efuse at production. Calibration code will automatically pull this value from efuse.
  - Medium scale: if you have purchased a batch of chips/modules or otherwise need to use the ones without V_REF programmed, you have two options:
    - Incorporate V_REF programming into your production flow. During production, you can either measure the V_REF using external tool (we have a function to pull V_REF out onto a pad), or you can use ESP32's ADC to measure a good-known voltage from a reference source, and use that to calculate V_REF; then burn it into efuse.
    - Add a reference voltage source onto your board and connect it to one of the ADC pins. Use that to calculate V_REF and feed the value into calibration code
  - Small scale: use a function which connects V_REF to a pad, measure V_REF voltage with a multimeter, write it into NVS or EFUSE.

V_REF compensation works pretty well, the errors which remain are due to run-time variation of V_REF. If you add an external reference voltage source and connect it to one of the ADC pins, you can apply correction at runtime, compensating even for temperature changes.

Linearization works best for low attenuation settings (0 and 1), as for these settings the transfer curve is closest to being linear. At attenuation setting 3 things get a bit more messy and hard to correct for, so we advise using lowest attenuation if possible. You will still not get rail-to-rail performance, but the result may be sufficient for a non-precision application.

(1): currently I can not give you any indication what that date might be.

imotion-software, 2017-08-22:
Hi @igrr, really thank you for your feedback!

We have 10 samples of a custom board with an external 1.2V voltage reference (TI LM4041A12) and operational amplifier to filter input signals and adapt impedance. They have an output range of 0.15V to 1.8V (in order to avoid start end end of scale). 

We tried to implement many of the linearization tables found (they are all different!) and also used an heavy software filter on the external voltage reference pad to correct dynamic fluctuations caused by the internal voltage reference instability and work always with the correct end of scale.

I tried this approach on 10 different boards, and with all of them we have acquisition errors close to the 5-10% (which is useless for an ADC).

In my opinion, the non-linear characteristic is something different chip by chip, it's not only a matter of the internal voltage reference. And if this true, there's absolutely no way to compensate it.

imotion-software, 2017-08-22:
@Spritetm did you try to plot non-linear characteristic on different ESP32 chips?

Few users tried to plot it and they all get different results. I personally tried on few samples and always get something different...

jan-bozelie, 2017-08-23:
No we have as utility a pilot development  for in the medium gas distribution grid ( ongoing)

   1 natural gas  presure sensor  0-300mBar gauge --> 5V suply  0-3.3V output per home ( the sensor is max 0.5% error and is ok)

      And also not sensitive for 0.5V voltage variations in the supply

There is no noticeble noice on the suply voltage during this experiments

 

The ESP board has a stabelizer 5V tot 3.3V  feeded with a usb adapter / or laptop usb port 

 

  

 We did make 10 samples ahead of a pilot of  5000pc

   For each  10 sensors we did make a voltage  measurement readung compared to a voltage calibrator , with oversampling and avagaging

 

From these measurements we did make a correction curve, reducing  10%relative eror to 1% we can live with that

 

Than we checked the other , assuming the correction curve is aplicapable for all esp

  But no -->  worst case error between    10 ESP32 samples is 20% , what would inplement  we would have to calibrate each sensor seperate

  What would take at lease 20min labour , and not afordable

 

Conclusion , we have to look for an other solution , or a fix

   One experiment we planed =  ( put more analog inputs in parallel of the same ESP32 )

jan-bozelie, 2017-08-23:
I can confirm that is is chip to chip different from our 10 samples 

 In my aplication , (one sensor one esp32),  i could live with an havy correction algoritm

I cant live with labour time and cost ,  for calibrating each analog input & each esp seperatly 

 

Is there a bad sample ranproduction range ??

Spritetm, 2017-08-23:
http://i.imgur.com/mTkip9V.png
This is a set of a few boards. (We have more, but this is data I have right here). As you can see, the reference voltage of all ESP32s is slightly different (hence the steepness of the curve differs from esp to esp) and the start and ends of each curve as well as the entire attn=3 curve is non-linear, but apart from the reference voltage, the ESPs do not really have that different-looking curves.

imotion-software, 2017-08-24:
Hi @Spritetm ! Is it possibile to have raw data of those 10 samples?

vseven, 2017-09-01:
Here is some testing I did tonight using freshly pulled code from repo and built firmware.  12-bit and atten at -11db (defaults).  I do not know enough to change the atten or width...I can hack away at Arduino sketches but I don't know if I can change it from there and I'm not sure where to change it in the source before I build the firmware.  If anyone can tell me I'll try the -6db setting instead of the default -11db and see if it's better:

[image]

[ESP32-Voltage-vs-ADC-Reading.txt](https://github.com/espressif/esp-idf/files/1269042/ESP32-Voltage-vs-ADC-Reading.txt)

imotion-software, 2017-09-01:
Hi @vseven,

if you want to change attenuation, there's a specific function "adc1_config_channel_atten".
For width, you can change it calling "adc1_config_width".

I think -6dB should have a more linear characteristic, but probably reducing width will simply reduce noise but not non-linearities.

If you make a test at -6dB, please share result because I'm interested too.

vseven, 2017-09-01:
Ive tried those commands in my Arduino sketch, multiple times, and just get a compile error.  I must not have the syntax correct nor can i find it anywhere.  I made a post on ESP32.com's forums asking but no reply yet.  Once i change it i will retest and post.

vseven, 2017-09-01:
Clarification:  I have tried "analogReadResolution" and "analogSetAttenuation" from Arduino and it fails to compile.  I guess I can try changing the static set values in esp-hal-adc.c and recompile the firmware?

androidcho, 2017-09-26:
Does this bug get fixed?

vseven, 2017-09-26:
No but i did a lot of work on getting better results that might help you: https://esp32.com/viewtopic.php?f=19&t=2881

negativekelvin, 2017-09-26:
https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/adc.html#adc-calibration

rojer, 2017-09-28:
ok, so basically vref needs to be routed to gpio, measured and recorded in the particular device so it can apply the right correction values. this is unfortunate. i assume this will be fixed in the future revisions of the chip?

in the mean time, i have an application where i only need to measure value of one thermistor. i wonder if i can use vref to gpio routing to avoid the extra calibration step.
say, i have a circuit like this:

```
IO25 (Vref) ---,
               |
               RT
               |
IO34 ----------*
               |
               RB
               |
              ---
               -
               .
```

`RT` is the thermistor, `RB` is a bias resistor, value TBD. will using `Vref` like this allow me to get accurate reading of drop across `RT` on IO34 without calibration? using attenuation of 0, naturally.

DrBomb, 2017-09-28:
How would we "burn" the Vref into the fuses? I'd like to make the test for every ESP32, store the value and not worrying about measuring it again.

negativekelvin, 2017-09-28:
@rojer they say chips will ship with factory cal burned to efuse at some unspecified time in the future or possibly by special order.

rojer, 2017-09-28:
@negativekelvin that makes sense.

@DrBomb i know you're a mOS user, so this may be handy - `mos` utility can operate on eFuses:
```
$ mos -X esp32-efuse-get
...
$ mos -X esp32-efuse-set
Error: one or more ops required. op is 'fuse=value', 'fuse=@file' or 'fuse.{WD|RD}=1'
```
i guess you could use (part of) the `user_key` field for this.

rojer, 2017-09-29:
FWIW, here are my results from 10 devices i had lying around, calculated error % before calibration and after.
columns are:
id, measured voltage on input: raw reading => voltage @ vref 1100 (error %); measured vref => voltage @ vref (error %).
```
00, 1.648: 1953 => 1.697 (3.0%); vref 1.070 => 1.657 (0.5%)
04, 1.669: 1886 => 1.642 (1.6%); vref 1.100 => 1.642 (1.6%)
05, 1.650: 1835 => 1.600 (3.1%); vref 1.130 => 1.639 (0.7%)
06, 1.653: 1856 => 1.618 (2.2%); vref 1.122 => 1.648 (0.3%)
AC, 1.651: 1819 => 1.586 (4.1%); vref 1.132 => 1.628 (1.4%)
03, 1.643: 1776 => 1.552 (5.8%); vref 1.143 => 1.608 (2.2%)
tt, 1.635: 1778 => 1.556 (5.1%); vref 1.129 => 1.595 (2.5%)
SL, 1.658: 1839 => 1.604 (3.4%); vref 1.119 => 1.629 (1.2%)
5C, 1.640: 1819 => 1.587 (3.3%); vref 1.129 => 1.625 (0.9%)
9C, 1.654: 1891 => 1.646 (0.5%); vref 1.105 => 1.652 (0.1%)
```
so, calibration gets the errors down, but doesn't quite eliminate them.

also, i found that reading ADC disables vref output to the pin, which is unfortunate.

negativekelvin, 2017-09-30:
Are you averaging multiple samples?

jan-bozelie, 2017-09-30:
Yes we did and it improved the noice error , as we could expect

jan-bozelie, 2017-09-30:
Thanks , for the info -->  informative

  however, calibrating each io is killing each business case

I did buy pressure sensors with 0.25% accuracy  , adding than 2.5% is a lot

rojer, 2017-09-30:
> Are you averaging multiple samples?
i was not. i'd take a reading every second, print it out, and then use one of these lines to compute.
```
[Sep 29 22:25:36.037] timer_cb             Tick 1890 1645 (vref 1100) 1651 (vref 1105)
[Sep 29 22:25:37.037] timer_cb             Tock 1887 1643 (vref 1100) 1649 (vref 1105)
[Sep 29 22:25:38.037] timer_cb             Tick 1886 1642 (vref 1100) 1648 (vref 1105)
[Sep 29 22:25:39.037] timer_cb             Tock 1891 1646 (vref 1100) 1652 (vref 1105)
[Sep 29 22:25:40.037] timer_cb             Tick 1894 1649 (vref 1100) 1655 (vref 1105)
[Sep 29 22:25:41.037] timer_cb             Tock 1888 1644 (vref 1100) 1650 (vref 1105)
```

> calibrating each io is killing each business case

exactly my thinking. it's imperative that Espressif starts doing it at some point, or we can forget about ADC for all but the most trivial applications.

androidcho, 2017-10-04:
So we won't get linearization data integrated into ESP-IDF? Very bad if that's the case..

projectgus, 2017-10-05:
> So we won't get linearization data integrated into ESP-IDF? Very bad if that's the case..

Linearization of ADC readings is integrated into ESP-IDF now via the [esp_adc_cal API](https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/adc.html#adc-calibration). The part which is still pending is being able to calibrate the chip's internal Vref (a fixed voltage offset applied to the linearized value) from a value shipped in the ESP32 from the factory.

Right now the options are:
* Use a default VRef value for all chips and accept that calibration results will be much more linear than "raw" readings but may be off by a fixed offset.
* Measure VRef on each ESP32 (as described above) and store this value for later use with the calibrated ADC API.

Future revisions of the ESP32 will have VRef measured as part of the in-factory test, with the value stored in efuse.

androidcho, 2017-10-05:
Well, that's some very good news! Thanks for the reply!

zhivko, 2017-10-15:
Would it be possible to make calibration routine using DAC and ADC? Both from esp32?

elFallino, 2017-11-19:
Should the characteristics returned by esp_adc_cal_get_characteristics() be valid for all channels of ADC1?

I have 234mV at Channel 7 (GPIO 35) and 3010mV at Channel 0 (GPIO36) (measured with a multimeter).
Resolution is set to 12Bit, attenuation is set to 11dB.

**Trial 1**
When I play around with the vref parameter passed to _esp_adc_cal_get_characteristics()_ so that the voltage at Channel 7 (GPIO35) measured correctly by the ESP32, the voltage for Channel 0 (GPIO 36) is off by ~+230mV.
The determined Vref value here is 1163mV.

**Trial 2**
When I do it vice versa, passing a Vref parameter to _esp_adc_cal_get_characteristics()_ so the voltage at Channel 0 (GPIO 35) is measured correctly by the EPS32, the voltage for Channel 7 (GPIO 35) is off by 
~-70mV.
The determined Vref value here is 1056mV.
When I connect GPIO35 with GPIO36 so that both do get the same voltage, measurements on both channels are equal.

So it seems that even with the linearisation tables in use, its not getting that good. What should be the the optimal voltage connected to an ADC Channel determine ones Vref voltage that way? Maybe 0.5xVmax?

In my case, the three Pins where I could route the internal Vref to for measurement are not available due to already connected components on my prototype board so I can not measure it directly (and I could not ask the customer to do so at all).

My initial idea was to connect a well-known voltage to a free ADC Channel, manipulating the Vref parameter in a loop until the measured value is equal to the well-known one and use that for all channels.

jan-bozelie, 2017-11-20:
@ESP32DE 
i will leave my finger from the ESP32 until the problem is hardware solved
to ESP if you cant do it right  , dont make such a product !!!

to the rest of the world , is there an other product out there we can use , and will do better 
this esp32 we will never get it working right ,

negativekelvin, 2017-11-20:
@elFallino if you are going to try that you should pick the value of vref that has the smallest max error at both voltages.

elFallino, 2017-11-20:
That's not easy to implement as it would require to switch between two stable, well-known voltages with hardware. That would also require to generate a second voltage.

One of the voltages I need to measure is the output of a hall sensor for current measurement with a resolution with 33mV/Ampere. So even some few mV's  error would be a big off in current measurement.

What a pain in the ass:(

rojer, 2017-12-12:
question about linearization. i see that 0 raw reading is not actually 0 in voltage after linearization, it's actually 110 mV. similarly, max raw reading (4095 in 12-bit mode) is interpolated to 3160 mV when vdd is set to 3.3. what is the reason for this?

negativekelvin, 2017-12-12:
@rojer check the graph posted above https://github.com/espressif/esp-idf/issues/164#issuecomment-326476297

tcsaba101, 2017-12-26:
Nonlinearity could be handled. The main problem is the ambiguity. I have a video where the input voltage stays within uV precise, while the reading swinging in a 1,6% range between 3746 to 3810. This ADC is unusable. And not forget I have put a 68nF capacitor on th input, but almost no effect. This looka serious mistake, I have to use a separate ADC for most of the analog projects with ESP32.

rajkovicbojan, 2018-02-08:
Hi people,
Have any solution for ambiguity? I am trying to measure voltage from current transformer via 22omh resistror. And I get always different measurment for input current of 0A. Can someone give me advice how to calibrate my ADC and get right measeurment?

Here is my peace of code for ADC initailization and for  measuring:

void init_adc(void)
{

	// ADC capture width is 12Bit
    adc1_config_width(ADC_WIDTH_12Bit);
	// ADC1 channel 6 is GPIO34, full-scale voltage 3.9V
    adc1_config_channel_atten(ADC1_CHANNEL_6,ADC_ATTEN_11db);
	// ADC1 channel 7 is GPIO35, full-scale voltage 3.9V
    adc1_config_channel_atten(ADC1_CHANNEL_7,ADC_ATTEN_11db);
	// ADC1 channel 4 is GPIO32, full-scale voltage 3.9V
    adc1_config_channel_atten(ADC1_CHANNEL_4,ADC_ATTEN_11db);

    // Calculate ADC characteristics i.e. gain and offset factors
    esp_adc_cal_get_characteristics(V_REF, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, &characteristics);
}
void ADC_Task(){

	results[0] = adc1_to_voltage(ADC1_CHANNEL_6,  &characteristics); // ADC1 channel 6 is GPIO34
	results[1] = adc1_to_voltage(ADC1_CHANNEL_7,  &characteristics); // ADC1 channel 7 is GPIO35
	results[2] = adc1_to_voltage(ADC1_CHANNEL_4,  &characteristics); // ADC1 channel 4 is GPIO32
}

[125 more replies]