espressif/esp-idf #165: WiFi Server

SirRibeiro, 2016-12-15:
Hello,
Is there any way of creating a http server?

Thanks

nkolban, 2016-12-15:
Howdy,
There are a number of techniques that folks have had success using.  Many of them have been discussed on the forums found here:

http://esp32.com/index.php

This web site is going to be a good site for your studies and is invaluable for most of us.  If you search for "http server" you will be rewarded.   Some of the ones I have enjoyed including using the "Mongoose" library, the Arduino libraries and the JavaScript engines to create a Node.js like server.

SirRibeiro, 2016-12-15:
Thank you very much for your info, i will look into it :)

projectgus, 2016-12-15:
I'm going to close this as it's probably a better topic for forum discussion, but good @SirRibeiro finding a good solution. :)

closed 2016-12-15 by projectgus