espressif/esp-idf #166: project.mk:279: *** multiple target patterns.  Stop.

krzychb, 2016-12-15:
After `git clone` of current esp-idf and attempting to `make menuconfig` of `hello_world_main.c` I am getting the following error:

```
$ make menuconfig
C:/msys32/esp-idf/make/project.mk:279: *** multiple target patterns.  Stop.
```

Replacing `project.mk` to an older version that I loaded yesterday resolves this issue.

Do I miss some extra configuration that should be done somewhere else?

projectgus, 2016-12-15:
Hmm, I'm not able to reproduce this with 87d0eb0fdb0. But the line in the error is something that was changed recently. It looks like the APP_ELF target is being generated with a colon embedded in the dependencies somewhere, which is invalid make syntax.

A few things to try, if you haven't already:
* Does a clean build help? (please move the "broken" build directory out of the way so we can investigate it if it turns out to be the problem.)
* If you grep recursively for COMPONENT_BUILD_DEPS in the build directory, do you see any names which might have an embedded colon character?
* If you add a $(warning) line above the line with the error. like this:
```
$(warning $(COMPONENT_LINKER_DEPS))
$(APP_ELF): $(foreach libcomp,$(COMPONENT_LIBRARIES),$(BUILD_DIR_BASE)/$(libcomp)/lib$(libcomp).a) $(COMPONENT_LINKER_DEPS)
```
What gets printed out, and does that contain any embedded colons? For comparison, mine is:
```
/home/gus/esp/32/idf/make/project.mk:279:   /home/gus/esp/32/idf/components/bt/lib/libbtdm_app.a  /home/gus/esp/32/idf/components/esp32/lib/libcore.a /home/gus/esp/32/idf/components/esp32/lib/libnet80211.a /home/gus/esp/32/idf/components/esp32/lib/libphy.a /home/gus/esp/32/idf/components/esp32/lib/librtc.a /home/gus/esp/32/idf/components/esp32/lib/libpp.a /home/gus/esp/32/idf/components/esp32/lib/libwpa.a /home/gus/esp/32/idf/components/esp32/lib/libsmartconfig.a /home/gus/esp/32/idf/components/esp32/lib/libcoexist.a /home/gus/esp/32/idf/components/esp32/lib/libwps.a /home/gus/esp/32/idf/components/esp32/lib/libwpa2.a /home/gus/esp/32/idf/components/esp32/ld/esp32.common.ld /home/gus/esp/32/idf/components/esp32/ld/esp32.rom.ld /home/gus/esp/32/idf/components/esp32/ld/esp32.peripherals.ld         /home/gus/esp/32/idf/components/newlib/lib/libc.a /home/gus/esp/32/idf/components/newlib/lib/libm.a
```
* I don't know if `make -p` will work correctly because of the parsing error, but if it does then somewhere in the enormous mountain of output there should be two lines like this for the .elf targets (grep for `.elf: `):
```
/home/gus/esp/32/idf/examples/01_hello_world/build/bootloader/bootloader.elf: /home/gus/esp/32/idf/examples/01_hello_world/build/bootloader/bootloader_s
upport/libbootloader_support.a /home/gus/esp/32/idf/examples/01_hello_world/build/bootloader/log/liblog.a /home/gus/esp/32/idf/examples/01_hello_world/b
uild/bootloader/spi_flash/libspi_flash.a /home/gus/esp/32/idf/examples/01_hello_world/build/bootloader/micro-ecc/libmicro-ecc.a /home/gus/esp/32/idf/exa
mples/01_hello_world/build/bootloader/main/libmain.a /home/gus/esp/32/idf/components/bootloader/src/main/esp32.bootloader.ld /home/gus/esp/32/idf/compon
ents/esp32/ld/esp32.rom.ld
```
and elsewhere
```
/home/gus/esp/32/idf/examples/01_hello_world/build/hello-world.elf: /home/gus/esp/32/idf/examples/01_hello_world/build/app_update/libapp_update.a /home/gus/esp/32/idf/examples/01_hello_world/build/bootloader_support/libbootloader_support.a /home/gus/esp/32/idf/examples/01_hello_world/build/bt/libbt.a /home/gus/esp/32/idf/examples/01_hello_world/build/driver/libdriver.a /home/gus/esp/32/idf/examples/01_hello_world/build/esp32/libesp32.a /home/gus/esp/32/idf/examples/01_hello_world/build/ethernet/libethernet.a /home/gus/esp/32/idf/examples/01_hello_world/build/expat/libexpat.a /home/gus/esp/32/idf/examples/01_hello_world/build/freertos/libfreertos.a /home/gus/esp/32/idf/examples/01_hello_world/build/json/libjson.a /home/gus/esp/32/idf/examples/01_hello_world/build/log/liblog.a /home/gus/esp/32/idf/examples/01_hello_world/build/lwip/liblwip.a /home/gus/esp/32/idf/examples/01_hello_world/build/mbedtls/libmbedtls.a /home/gus/esp/32/idf/examples/01_hello_world/build/micro-ecc/libmicro-ecc.a /home/gus/esp/32/idf/examples/01_hello_world/build/newlib/libnewlib.a /home/gus/esp/32/idf/examples/01_hello_world/build/nghttp/libnghttp.a /home/gus/esp/32/idf/examples/01_hello_world/build/nvs_flash/libnvs_flash.a /home/gus/esp/32/idf/examples/01_hello_world/build/openssl/libopenssl.a /home/gus/esp/32/idf/examples/01_hello_world/build/spi_flash/libspi_flash.a /home/gus/esp/32/idf/examples/01_hello_world/build/tcpip_adapter/libtcpip_adapter.a /home/gus/esp/32/idf/examples/01_hello_world/build/ulp/libulp.a /home/gus/esp/32/idf/examples/01_hello_world/build/vfs/libvfs.a /home/gus/esp/32/idf/examples/01_hello_world/build/wpa_supplicant/libwpa_supplicant.a /home/gus/esp/32/idf/examples/01_hello_world/build/xtensa-debug-module/libxtensa-debug-module.a /home/gus/esp/32/idf/examples/01_hello_world/build/main/libmain.a /home/gus/esp/32/idf/components/bt/lib/libbtdm_app.a /home/gus/esp/32/idf/components/esp32/lib/libcore.a /home/gus/esp/32/idf/components/esp32/lib/libnet80211.a /home/gus/esp/32/idf/components/esp32/lib/libphy.a /home/gus/esp/32/idf/components/esp32/lib/librtc.a /home/gus/esp/32/idf/components/esp32/lib/libpp.a /home/gus/esp/32/idf/components/esp32/lib/libwpa.a /home/gus/esp/32/idf/components/esp32/lib/libsmartconfig.a /home/gus/esp/32/idf/components/esp32/lib/libcoexist.a /home/gus/esp/32/idf/components/esp32/lib/libwps.a /home/gus/esp/32/idf/components/esp32/lib/libwpa2.a /home/gus/esp/32/idf/components/esp32/ld/esp32.common.ld /home/gus/esp/32/idf/components/esp32/ld/esp32.rom.ld /home/gus/esp/32/idf/components/esp32/ld/esp32.peripherals.ld /home/gus/esp/32/idf/components/newlib/lib/libc.a /home/gus/esp/32/idf/components/newlib/lib/libm.a
```

(It's a little sobering seeing how long these lists are getting!)

Do any of those outputs have a sneaky colon character in them?

krzychb, 2016-12-16:
Hi @projectgus

Wow - thank you for extensive troubleshooting instructions with several steps ahead. 
Here is what I found:

> Does a clean build help? (please move the "broken" build directory out of the way so we can investigate it if it turns out to be the problem.)

```
Krzysztof@tdk-kmb-op780 MSYS /esp/01_hello_world
$ make clean
C:/msys32/esp-idf/make/project.mk:279: *** multiple target patterns.  Stop.
```

If I delete `build` directory and `sdkconfig` file, then I get
```
Krzysztof@tdk-kmb-op780 MSYS /esp/01_hello_world
$ make menuconfig
MENUCONFIG

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

C:/msys32/esp-idf/make/project.mk:279: *** multiple target patterns.  Stop.
```

> If you grep recursively for COMPONENT_BUILD_DEPS in the build directory, do you see any names which might have an embedded colon character?

I am not sure what should I do about `COMPONENT_BUILD_DEPS`. Here is waht I did. 

```
Krzysztof@tdk-kmb-op780 MSYS /esp/01_hello_world/build
$ grep -r ":" .
./app_update/component_project_vars.mk:app_update-build:
./bootloader_support/component_project_vars.mk:bootloader_support-build:
./bt/component_project_vars.mk:bt-build:
./driver/component_project_vars.mk:driver-build:
./esp32/component_project_vars.mk:esp32-build:
./ethernet/component_project_vars.mk:ethernet-build:
./expat/component_project_vars.mk:expat-build:
./freertos/component_project_vars.mk:freertos-build:
./json/component_project_vars.mk:json-build:
./log/component_project_vars.mk:log-build:
./lwip/component_project_vars.mk:lwip-build:
./main/component_project_vars.mk:main-build:
./mbedtls/component_project_vars.mk:mbedtls-build:
./micro-ecc/component_project_vars.mk:micro-ecc-build:
./newlib/component_project_vars.mk:newlib-build:
./nghttp/component_project_vars.mk:nghttp-build:
./nvs_flash/component_project_vars.mk:nvs_flash-build:
./openssl/component_project_vars.mk:openssl-build:
./spi_flash/component_project_vars.mk:spi_flash-build:
./tcpip_adapter/component_project_vars.mk:tcpip_adapter-build:
./ulp/component_project_vars.mk:ulp-build:
./vfs/component_project_vars.mk:vfs-build:
./wpa_supplicant/component_project_vars.mk:wpa_supplicant-build:
./xtensa-debug-module/component_project_vars.mk:xtensa-debug-module-build:
```

> If you add a $(warning) line above the line with the error. like this:

To get it run I fist need to delete `build` directory and `sdkconfig` file. After doing so I see the following:

```
Krzysztof@tdk-kmb-op780 MSYS /esp/01_hello_world
$ make menuconfig
C:/msys32/esp-idf/make/project.mk:279:
MENUCONFIG
configuration written to /esp/01_hello_world/sdkconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

C:/msys32/esp-idf/make/project.mk:279:   C:/msys32/esp-idf/components/bt/C:/msys32/esp-idf/components/bt/lib/libbtdm_app.a  C:/msys32/esp-idf/components/esp32/C:/msys32/esp-idf/components/esp32/lib/libcore.a C:/msys32/esp-idf/components/esp32/C:/msys32/esp-idf/components/esp32/lib/libnet80211.a C:/msys32/esp-idf/components/esp32/C:/msys32/esp-idf/components/esp32/lib/libphy.a C:/msys32/esp-idf/components/esp32/C:/msys32/esp-idf/components/esp32/lib/librtc.a C:/msys32/esp-idf/components/esp32/C:/msys32/esp-idf/components/esp32/lib/libpp.a C:/msys32/esp-idf/components/esp32/C:/msys32/esp-idf/components/esp32/lib/libwpa.a C:/msys32/esp-idf/components/esp32/C:/msys32/esp-idf/components/esp32/lib/libsmartconfig.a C:/msys32/esp-idf/components/esp32/C:/msys32/esp-idf/components/esp32/lib/libcoexist.a C:/msys32/esp-idf/components/esp32/C:/msys32/esp-idf/components/esp32/lib/libwps.a C:/msys32/esp-idf/components/esp32/C:/msys32/esp-idf/components/esp32/lib/libwpa2.a C:/msys32/esp-idf/components/esp32/ld/esp32.common.ld C:/msys32/esp-idf/components/esp32/ld/esp32.rom.ld C:/msys32/esp-idf/components/esp32/ld/esp32.peripherals.ld         C:/msys32/esp-idf/components/newlib/C:/msys32/esp-idf/components/newlib/lib/libc.a C:/msys32/esp-idf/components/newlib/C:/msys32/esp-idf/components/newlib/lib/libm.a
C:/msys32/esp-idf/make/project.mk:280: *** multiple target patterns.  Stop.
```

> What gets printed out, and does that contain any embedded colons? For comparison, mine is:

Comparing to your output mine has embedded colons `C:`.

> I don't know if make -p will work correctly because of the parsing error, but if it does then somewhere in the enormous mountain of output there should be two lines like this for the .elf targets (grep for .elf:):

I do not see `make -p` doing any compilation.

projectgus, 2016-12-16:
Ah, your Windows paths have colons in them as part of C:/. Of course they do! So the dependency paths are confusing make.

If you set `IDF_PATH=/c/msys32/esp-idf/` then this should work as-is, but I'll find a way to fix this properly in the build system.

krzychb, 2016-12-16:
Thanks, @projectgus!
Changing IDF_PATH resolves this issue.

projectgus, 2016-12-18:
Glad this sorted it out.

I'm going to leave this open, because the Windows instructions currently say to use C:/blah/ as paths, so we need to resolve this one way or another.

krzychb, 2016-12-19:
I have updated path in file inside of `C:/msys32/etc/profile.d/` and forgot the issue ever existed.
If this is the solution you lean towards, I can correct docs if this helps.

projectgus, 2016-12-19:
Thanks. I'm not a Windows user, but I feel like Windows users will prefer it if they can use a familiar syntax for paths (although it's already bit odd with forward-slashes, so I don't know.)

I'll take a look at how hard it is to fix in another way first, otherwise we'll change the instructions.

betzah, 2016-12-21:
I usually never post comments, however I just wanted to say THANK YOU for posting the solution: I've just spent hours with this exact same bug and also fixed it by changing the IDF_PATH.

cjsm74, 2016-12-25:
Sanitizing this line from windows-unfriendly whitespace characters solved it for me. The colon in the IDF_PATH doesn't seem to be a problem now.

projectgus, 2016-12-27:
Thanks for the update @cjsm74 . This should be fixed in esp-idf master now.

krzychb, 2017-01-03:
Hi @projectgus,

With the `master` branch of 2407faf6c6429bce90ed42e51cae531929e0697a, after running `make menuconfig` for `esp-idf-template` project, I see the following:

```
$ make menuconfig
c:/msys32/esp-idf/make/project.mk:63: *** If IDF_PATH is overriden on command line, it must be an absolute path with no embedded shell special characters.  Stop.
```
After changing above mentioned `export IDF_PATH="c:/msys32/esp-idf"` back to `export IDF_PATH="/c/msys32/esp-idf"`, the problem is gone.

I am running esp-idf on Windows 7. I did this test just before going to checkout branch `bugfix/eclipse_windows_paths` to do testing asked in https://github.com/espressif/esp-idf/pull/157#issuecomment-270048034

projectgus, 2017-01-04:
Thanks for updating, @krzychb . I'm not able to reproduce this on my Windows 7 system, on 2407faf if I run

```
export IDF_PATH="c:/esp/esp-idf"
make
```

... it works fine. Same if I set this environment variable value in Eclipse.

I'm using the msys shell from the pre-built environment (launched from Win+R as `c:\msys32\msys2_shell.cmd`). Is this the same environment as you? It's possible different shells may trigger different behaviour. The Windows 7 I am using is 32-bit, although I can't imagine why that would make a difference.

projectgus, 2017-01-04:
Well, this is interesting... Did a clean clone at the *exact* same path you were using:

```
$ cd /c/msys32/
$ git clone --recursive https://github.com/espressif/esp-idf.git
$ cd esp-idf
$ export IDF_PATH="c:/msys32/esp-idf"
$ cd examples/01_hello_world/
$ make
c:/msys32/esp-idf/make/project.mk:63: *** If IDF_PATH is overriden on command line, it must be an absolute path with no embedded shell special characters.  Stop.
```

It must be related to running under the MSYS2 directory.

projectgus, 2017-01-04:
This seems like it might be an MSYS bug

`$(wildcard "c:/msys32/esp-idf")` -> `"/esp-idf"`
`$(wildcard /esp-idf)` -> `""`

projectgus, 2017-01-04:
I've pushed another commit to the bugfix/eclipse_windows_paths branch, when you get a chance can you try it out please?

krzychb, 2017-01-08:
Hi @projectgus,

Here are my test results:

Make `01_hello_world` example from MSYS using:
- `export IDF_PATH="/c/msys32/esp-idf/"` - OK
- `export IDF_PATH="C:/msys32/esp-idf/"` - OK

Build  `01_hello_world` example from Eclipse using:
- `IDF_PATH` variable set to `/c/msys32/esp-idf/` - OK
- `IDF_PATH` variable set to `C:/msys32/esp-idf/` - failure:
```
/esp-idf/make/project.mk:62: *** IDF_PATH variable is not set to a valid directory..  Stop.
make: *** [/esp-idf/components/bootloader/Makefile.projbuild:31: /esp/01_hello_world/build/bootloader/bootloader.bin] Error 2
```
Tough cookie :smile:

krzychb, 2017-02-22:
Resolved in https://github.com/espressif/esp-idf/commit/8dede8f8a46f42661665e9b924b7f455d06ef00a

closed 2017-02-22 by krzychb

Paiklabs, 2018-12-19:
I solved the problem.
[https://www.esp32.com/viewtopic.php?f=13&t=8523](url)

guepy, 2019-08-27:
Hi , I'm facing error *IDF_PATH is not set to a valid directory* and my IDF_PATH is set to "C:/msys32/home/NDETTO FRAICHINNELLE/esp/esp-idf".
I suspect the occurence of space character in the username between NDETTO and FRAICHINNELLE, if so , plz to tell me how to manage.
Kind regards

guepy, 2019-08-27:
Hi , I'm facing error ***IDF_PATH is not set to a valid directory*** and my IDF_PATH is set to "C:/msys32/home/NDETTO FRAICHINNELLE/esp/esp-idf".
I suspect the occurence of space character in the username between NDETTO and FRAICHINNELLE, if so , plz to tell me how to manage.
Kind regards

renzbagaporo, 2019-08-27:
@guepy We mention in the docs that we don't support spaces in paths:

https://docs.espressif.com/projects/esp-idf/en/latest/get-started/#step-2-get-esp-idf

I'm afraid there's no other solution other than to move your ESP-IDF direcotry somewhere else to satisfy this constraint.

guepy, 2019-08-29:
Thx @renzbagaporo. I solved it using esp idf installer tools on your link