espressif/esp-idf #167: Buffer overflow on GATT BLE

jaracil, 2016-12-16:
Hi.

When I send a message to GATT attribute (using gatt_server example on ESP32) with payload greater than 20 bytes, it causes core dump.

Console drops this core dump:
```
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:

PC      : 0x400f2ec8  PS      : 0x00060c30  A0      : 0x800f1f00  A1      : 0x3ffd62e0  
A2      : 0x3ffc436c  A3      : 0x00000017  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000003  A7      : 0x00000000  A8      : 0x00000002  A9      : 0x00000000  
A10     : 0x00000001  A11     : 0x00060c23  A12     : 0x00060c23  A13     : 0xb33f0000  
A14     : 0xb33fffff  A15     : 0x00000001  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400f2ec8:0x3ffd62e0 0x400f1f00:0x3ffd6310 0x400f42fc:0x3ffd6330 0x40111720:0x3ffd6370 0x40110215:0x3ffd6390 0x401128e4:0x3ffd63b0 0x400ef577:0x3ffd63d0 0x40134bc2:0x3ffd63f0 0x400ef6e7:0x3ffd6410
```

On linux (Bluez tools) it can be reproduced with this commands:
(Replace mac with your ESP32 bt mac)

```
#gatttool -I -b 24:0A:C4:00:B7:60

[24:0A:C4:00:B7:60][LE]> connect
Connection successful
[24:0A:C4:00:B7:60][LE]> char-desc
handle: 0x0001, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x0002, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0003, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x0014, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x0015, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0016, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0017, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0018, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0019, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x001a, uuid: 00002aa6-0000-1000-8000-00805f9b34fb
handle: 0x0028, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x0029, uuid: 00002803-0000-1000-8000-00805f9b34fb
>> handle: 0x002a, uuid: 0000ff01-0000-1000-8000-00805f9b34fb <<
handle: 0x002b, uuid: 00002902-0000-1000-8000-00805f9b34fb

char-write-req 0x002A 0102030405060708090A0B0C0D0E0F101112131415 

```

xcguang, 2017-04-11:
Hi @jaracil ,
Please see https://github.com/espressif/esp-idf/issues/418 .

FayeY, 2017-05-04:
Hi @jaracil, is this problem solved? Can we close this issue?

jaracil, 2017-05-04:
Yes. Now It works fine.

Thanks.

closed 2017-05-04 by jaracil