espressif/esp-idf #168: RTC Watchdog triggered after reset

lollizze, 2016-12-16:
Hi
When I press reset, i always encounter a RTCWDT_RTC_RESET, what reason could it be to get this kind of reset, i have the wroom dev kit with the 1.0 release of the sdk

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3ffc0008,len:0
load:0x3ffc0008,len:20
ho 0 tail 12 room 4
load:0x40078000,len:2680
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034

Could be connected to #113 but i don't get the read error and it happens all the time

Spritetm, 2016-12-16:
It is a silicon error, as documented here: http://espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf

Apart from slowing boot times slightly, it is harmless. It will be fixed in the new revision of the ESP32 chip, which is expected to come out around february.

lollizze, 2016-12-16:
Thanks to the quick reply! That explains it, just an inconvenience that you can't get the real reset reason or is there an work around to get that?

igrr, 2016-12-16:
This happens only after power-on reset and after deep sleep reset. For deep sleep reset, we have a workaround which prevents this (so you get correct reset behavior). For the power-on reset, unfortunately there is no workaround, so you indeed get incorrect reset reason.

closed 2016-12-22 by igrr

Kashyapkoshti, 2020-11-19:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

I am getting this error in my esp32 devkit v1
How to solve it?
I am using arduino IDE for programming the esp32.