espressif/esp-idf #169: Exception at lwip/api/sockets.c:773

nkolban, 2016-12-16:
I am working on a large project that has lots of networking running in it.  I am obtaining a non-deterministic crash
that I can recreate 50% of the time, the other 50% it works as desired.  When the crash occurs, the following is the
stack dump:

```
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40130b76  PS      : 0x00060830  A0      : 0x80131cdc  A1      : 0x3ffcd380
A2      : 0x00000007  A3      : 0x3ffcd410  A4      : 0x00000002  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80130b69  A9      : 0x3ffcd360
A10     : 0x0000b4e7  A11     : 0x3ffcd384  A12     : 0x3ffcd398  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x3ffbeb58  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000002  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x40130b76:0x3ffcd380 0x40131cdc:0x3ffcd3e0 0x401064c0:0x3ffcd400

                                                                            Entering gdb stub now.
$T0b#e6
```
And attaching gdb and listing the backtrace shows ...

```
Reading symbols from ./build/app-template.elf...done.
Remote debugging using /dev/ttyUSB0
lwip_accept (s=<optimized out>, addr=0x3ffcd410, addrlen=0x2)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/api/sockets.c:773
773	    if (*addrlen > tempaddr.sa.sa_len) {
(gdb) bt
#0  lwip_accept (s=<optimized out>, addr=0x3ffcd410, addrlen=0x2)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/api/sockets.c:773
#1  0x40131cdc in lwip_accept_r (s=0, addr=0x3ffcd410, addrlen=0x3ffcd430)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/api/sockets.c:3183
#2  0x401064c0 in socket_server (ignore=<optimized out>)
    at /home/kolban/esp32/esptest/apps/workspace/duktape/main/./socket_server.c:51
(gdb) 
```

Within the logic of my application I have a task that is blocked on a sockets `accept()` call.
In another task I have a large library that is a JavaScript engine (Duktape) and we are polling
other sockets using "select".  When I use "nc" to send in some data, that is when it either
fails or doesn't.

My user code that is failing is:

```
	struct sockaddr_in clientAddress;

	while (1) {
		// Listen for a new client connection.
		ESP_LOGD(tag, "Waiting for new client connection");
		socklen_t clientAddressLength = sizeof(clientAddress);
		int clientSock = accept(sock, (struct sockaddr *)&clientAddress, &clientAddressLength);  // <!-- socket_server.c - Line 51
		if (clientSock < 0) {
			ESP_LOGE(tag, "accept: %d %s", clientSock, strerror(errno));
			goto END;
		}
```

I will continue to attempt to diagnose and any suggestions on tests to run would be welcomed.  I am running the very latest ESP-IDF.
I'm also raising this issue in case others encounter similar.  Exceptions that occur within ESP-IDF (or LWIP) when passed in what
are believed to be good parameters are where I turn to Github and issues.

negativekelvin, 2016-12-18:
lwip_accept_r (s=0, addr=0x3ffcd410, **addrlen=0x3ffcd430**)
lwip_accept (s=<optimized out>, addr=0x3ffcd410, **addrlen=0x2**)

Not sure why the argument value changed

projectgus, 2016-12-22:
If you allocate a larger stack size to the task which is crashing, does this help?

nkolban, 2016-12-22:
Unfortunately I don't have any recreate steps.  As such, if we want to close then by all means.  The issue should still show up in searches at Github.  The thinking is that if one sees an exception in ESP-IDF supplied code, one could search in the issues and see if it has been seen before.  That way we may see patterns in reproduction or else learn that there is something "endemic" somewhere that might need attention.

projectgus, 2016-12-22:
I see where you're coming from, and I agree entirely that this is potentially useful if other people find it this way.

From my point of view as an esp-idf developer, there's not a lot of value to keeping issues open when there's no way to tell if the issue is in esp-idf or is caused by some combination of external code. 

So my preference is to close unreproducable issues for now. If anyone else comes across the same crash then they can reopen then, and we can look for steps to reproduce.

I'll close this one now. If you think any of your similar issues belong in the same category (unable to reproduce, or to determine if caused by esp-idf or some external factor) please close these also.

closed 2016-12-22 by projectgus