espressif/esp-idf #171: Exception at LWIP igmp.c Line 676

nkolban, 2016-12-18:
I have started receiving an exception.   It appears to be originating within LWIP logic.  I am raising this issue as a record of the story in the hopes that if others see it, it may spark something further:

```
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400ef56b  PS      : 0x00060930  A0      : 0x800ea5da  A1      : 0x3ffc6090  
A2      : 0x8011884d  A3      : 0x3ffc36d4  A4      : 0x00000001  A5      : 0x3ffc58b8  
A6      : 0x3ffc588c  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffc6070  
A10     : 0x3ffc36d4  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3ffc5994  A15     : 0x3ffc5954  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x8011885b  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400ef56b:0x3ffc6090 0x400ea5da:0x3ffc60b0 0x400ea674:0x3ffc60d0 0x400ed06f:0x3ffc6100
```

And then attached GDB:

```
Remote debugging using /dev/ttyUSB0
igmp_tmr () at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/ipv4/igmp.c:676
676	    if (group->timer > 0) {
(gdb) bt
#0  igmp_tmr () at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/ipv4/igmp.c:676
#1  0x400ea5da in igmp_timer (arg=0x0) at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/timers.c:210
#2  0x400ea674 in sys_timeouts_mbox_fetch (mbox=0x3ffbd750 <mbox>, msg=0x3ffc6100)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/timers.c:575
#3  0x400ed06f in tcpip_thread (arg=<optimized out>)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/api/tcpip.c:101
(gdb) 
```

projectgus, 2016-12-22:
Do you have any steps to reproduce this exception?

nkolban, 2016-12-22:
Unfortunately I don't have any recreate steps.  As such, if we want to close then by all means.  The issue should still show up in searches at Github.  The thinking is that if one sees an exception in ESP-IDF supplied code, one could search in the issues and see if it has been seen before.  That way we may see patterns in reproduction or else learn that there is something "endemic" somewhere that might need attention.

nkolban, 2016-12-22:
Closed for just now.

closed 2016-12-22 by nkolban