espressif/esp-idf #173: 06_sntp reset after reboot

hdavidka, 2016-12-19:
Hi,

After the deepsleep mode, I get the following error:

```
C:/ESP32/esp-idf/components/freertos/heap_regions.c:531 (vPortDefineHeapRegionsTagged)- assert failed!

Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)

Register dump:

PC      : 0x40009203  PS      : 0x00060736  A0      : 0x00000000  A1      : 0x3ffe3bf0

A2      : 0x00000067  A3      : 0x00000067  A4      : 0x00000010  A5      : 0xffffffff

A6      : 0x00000000  A7      : 0xffffffe3  A8      : 0x00000000  A9      : 0x3ffe3b60

A10     : 0x00000000  A11     : 0x3f40437c  A12     : 0x3ffe3b2f  A13     : 0x00000031

A14     : 0x00000000  A15     : 0x3ffe3b34  SAR     : 0x00000004  EXCCAUSE: 0x00000001

EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x40009203:0x3ffe3bf0 0x00000000:0x3ffe3c10 0x400d1b9a:0x3ffe3c30 0x40080abe:0x3ffe3c60 0x400784b5:0x3ffe3ca0 0x4007867d:0x3ffe3cd0 0x40078a54:0x3ffe3d50 0x40080101:0x3ffe3e70 0x40007c34:0x3ffe3eb0 0x40000740:0x3ffe3f20

Rebooting...
```

After the first reset, it can't sync with the servers anymore. The ESP32 doesn't save the time and it print:
`I (2213) example: Boot count: 0`
Which is 1 after a fresh reset.

hdavidka, 2016-12-19:
With V1.0 it's working fine.

projectgus, 2016-12-22:
Thanks for reporting this @hdavidka. I'm not able to reproduce this with the current master branch (commit cca883490). Could you please do a clean build with this commit and try again?

projectgus, 2016-12-28:
Closing for now, but please reopen if the problem is not fixed.

closed 2016-12-28 by projectgus