espressif/esp-idf #174: Make get_time_since_boot public

rojer, 2016-12-20:
[`get_time_since_boot`](https://github.com/espressif/esp-idf/blob/655fd29/components/newlib/time.c#L126) provides a high granularity low overhead monotonic counter, which is a useful thing to have.
it is used by `gettimeofday` but would be useful to have on its own right.
would be nice to have it made public.
my own use case for it: busy-loop short delays.

igrr, 2018-02-12:
Sorry, forgot to update when this was added: the microsecond timer is now exposed via [`esp_timer_get_time`](http://esp-idf.readthedocs.io/en/latest/api-reference/system/esp_timer.html#_CPPv218esp_timer_get_timev) function.

closed 2018-02-12 by igrr

rojer, 2018-02-12:
great, thanks!

rustyx, 2019-03-16:
@igrr Unfortunately `esp_timer_get_time` is not available in ESP8266 SDK, but `get_time_since_boot` is available in both. So, the latter is more portable atm.

Beliwars, 2019-11-13:
> Sorry, forgot to update when this was added: the microsecond timer is now exposed via [`esp_timer_get_time`](http://esp-idf.readthedocs.io/en/latest/api-reference/system/esp_timer.html#_CPPv218esp_timer_get_timev) function.

Hello @igrr, I wanted to jump here to ask if there is any way to modify the value returned by esp_timer_get_time() function; it's for handling some timing alignment after coming back from a deepsleep.