espressif/esp-idf #175: undefined reference to `wcsftime' error in C++

imxieyi, 2016-12-20:
I wrote a program in C++ and it uses sstream library:
```
#include <sstream>
using namespace std;
extern "C"{
	void app_main(){
		ostringstream os;
	}
}
```
In order to compile it, I also changed the Makefile into:
```
#
# This is a project Makefile. It is assumed the directory this Makefile resides in is a
# project subdirectory.
#

PROJECT_NAME := hello-world

include $(IDF_PATH)/make/project.mk

COMPONENT_LDFLAGS += -lstdc++ --verbose
```
When I make the project, I got the following errors:
```
LD hello-world.elf
Using built-in specs.
COLLECT_GCC=C:\msys32\opt\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe
COLLECT_LTO_WRAPPER=c:/msys32/opt/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/4.8.5/lto-wrapper.exe
Target: xtensa-esp32-elf
Configured with: /home/gus/esp/crosstool-NG/.build/src/gcc-4.8.5/configure --build=x86_64-build_pc-linux-gnu --host=i686-host_pc-mingw32 --target=xtensa-esp32-elf --prefix=/home/gus/esp/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/home/gus/esp/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-sysroot=/home/gus/esp/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG 8d95cad' --disable-__cxa_atexit --enable-cxx-flags='-fno-exceptions -fno-rtti' --with-gmp=/home/gus/esp/crosstool-NG/.build/xtensa-esp32-elf/buildtools/complibs-host --with-mpfr=/home/gus/esp/crosstool-NG/.build/xtensa-esp32-elf/buildtools/complibs-host --with-mpc=/home/gus/esp/crosstool-NG/.build/xtensa-esp32-elf/buildtools/complibs-host --with-isl=/home/gus/esp/crosstool-NG/.build/xtensa-esp32-elf/buildtools/complibs-host --with-cloog=/home/gus/esp/crosstool-NG/.build/xtensa-esp32-elf/buildtools/complibs-host --with-libelf=/home/gus/esp/crosstool-NG/.build/xtensa-esp32-elf/buildtools/complibs-host --enable-lto --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++ -lm' --enable-target-optspace --without-long-double-128 --disable-libgomp --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-nls --disable-multilib --enable-languages=c,c++
Thread model: single
gcc version 4.8.5 (crosstool-NG 8d95cad)
COMPILER_PATH=c:/msys32/opt/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/4.8.5/;c:/msys32/opt/xtensa-esp32-elf/bin/../libexec/gcc/;c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/bin/
LIBRARY_PATH=c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/;c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/;c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/lib/;c:/msys32/opt/xtensa-esp32-elf/bin/../xtensa-esp32-elf/sysroot/lib/;c:/msys32/opt/xtensa-esp32-elf/bin/../xtensa-esp32-elf/sysroot/usr/lib/
COLLECT_GCC_OPTIONS='-nostdlib' '-LD:/01_hello_world/build/app_update' '-LD:/01_hello_world/build/bootloader' '-LD:/01_hello_world/build/bootloader_support' '-LD:/01_hello_world/build/bt' '-LD:/01_hello_world/build/driver' '-LD:/01_hello_world/build/esp32' '-LD:/01_hello_world/build/esptool_py' '-LD:/01_hello_world/build/ethernet' '-LD:/01_hello_world/build/expat' '-LD:/01_hello_world/build/freertos' '-LD:/01_hello_world/build/idf_test' '-LD:/01_hello_world/build/json' '-LD:/01_hello_world/build/log' '-LD:/01_hello_world/build/lwip' '-LD:/01_hello_world/build/mbedtls' '-LD:/01_hello_world/build/micro-ecc' '-LD:/01_hello_world/build/newlib' '-LD:/01_hello_world/build/nghttp' '-LD:/01_hello_world/build/nvs_flash' '-LD:/01_hello_world/build/openssl' '-LD:/01_hello_world/build/partition_table' '-LD:/01_hello_world/build/spi_flash' '-LD:/01_hello_world/build/tcpip_adapter' '-LD:/01_hello_world/build/ulp' '-LD:/01_hello_world/build/vfs' '-LD:/01_hello_world/build/wpa_supplicant' '-LD:/01_hello_world/build/xtensa-debug-module' '-LD:/01_hello_world/build/main' '-u' 'call_user_start_cpu0' '-LD:/esp32/esp-idf/components/bt/lib' '-LD:/esp32/esp-idf/components/esp32/lib' '-LD:/esp32/esp-idf/components/esp32/ld' '-T' 'esp32_out.ld' '-T' 'esp32.common.ld' '-T' 'esp32.rom.ld' '-T' 'esp32.peripherals.ld' '-v' '-o' 'D:/01_hello_world/build/hello-world.elf'
 c:/msys32/opt/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/4.8.5/collect2.exe --sysroot=c:\msys32\opt\xtensa-esp32-elf\bin\../xtensa-esp32-elf/sysroot -o D:/01_hello_world/build/hello-world.elf -u call_user_start_cpu0 -LD:/01_hello_world/build/app_update -LD:/01_hello_world/build/bootloader -LD:/01_hello_world/build/bootloader_support -LD:/01_hello_world/build/bt -LD:/01_hello_world/build/driver -LD:/01_hello_world/build/esp32 -LD:/01_hello_world/build/esptool_py -LD:/01_hello_world/build/ethernet -LD:/01_hello_world/build/expat -LD:/01_hello_world/build/freertos -LD:/01_hello_world/build/idf_test -LD:/01_hello_world/build/json -LD:/01_hello_world/build/log -LD:/01_hello_world/build/lwip -LD:/01_hello_world/build/mbedtls -LD:/01_hello_world/build/micro-ecc -LD:/01_hello_world/build/newlib -LD:/01_hello_world/build/nghttp -LD:/01_hello_world/build/nvs_flash -LD:/01_hello_world/build/openssl -LD:/01_hello_world/build/partition_table -LD:/01_hello_world/build/spi_flash -LD:/01_hello_world/build/tcpip_adapter -LD:/01_hello_world/build/ulp -LD:/01_hello_world/build/vfs -LD:/01_hello_world/build/wpa_supplicant -LD:/01_hello_world/build/xtensa-debug-module -LD:/01_hello_world/build/main -LD:/esp32/esp-idf/components/bt/lib -LD:/esp32/esp-idf/components/esp32/lib -LD:/esp32/esp-idf/components/esp32/ld -Lc:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5 -Lc:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc -Lc:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/lib -Lc:/msys32/opt/xtensa-esp32-elf/bin/../xtensa-esp32-elf/sysroot/lib -Lc:/msys32/opt/xtensa-esp32-elf/bin/../xtensa-esp32-elf/sysroot/usr/lib --gc-sections -static --start-group -lapp_update -lbootloader_support -lbt -lbtdm_app -ldriver -lesp32 D:/esp32/esp-idf/components/esp32/libhal.a -lcore -lnet80211 -lphy -lrtc -lpp -lwpa -lsmartconfig -lcoexist -lwps -lwpa2 -lethernet -lexpat -lfreertos --undefined=uxTopUsedPriority -ljson -llog -llwip -lmbedtls -lmicro-ecc D:/esp32/esp-idf/components/newlib/lib/libc.a D:/esp32/esp-idf/components/newlib/lib/libm.a -lnewlib -lnghttp -lnvs_flash -lopenssl -lspi_flash -ltcpip_adapter -lulp -lvfs -lwpa_supplicant -lxtensa-debug-module -lmain -lstdc++ -lgcc --end-group -EL -Map=D:/01_hello_world/build/hello-world.map -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/lib\libstdc++.a(time_members.o):(.literal._ZNKSt11__timepunctIwE6_M_putEPwjPKwPK2tm+0x0): undefined reference to `wcsftime'
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/lib\libstdc++.a(time_members.o): In function `std::__timepunct<wchar_t>::_M_put(wchar_t*, unsigned int, wchar_t const*, tm const*) const':
/home/gus/esp/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/src/c++98/time_members.cc:136: undefined reference to `wcsftime'
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/lib\libstdc++.a(vterminate.o):(.literal._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x8): undefined reference to `_impure_ptr'
collect2.exe: error: ld returned 1 exit status
make: *** [/d/esp32/esp-idf/make/project.mk:286：/d/01_hello_world/build/hello-world.elf] Error 1
```
If I delete "ostringstream os;" in the program, it can successfully compile.
I tried to remove all -nostdlib in project.mk, but it only got me with more errors.
So how can I resolve it? I guess that there should be some C++ libraries in components/newlib.

Spritetm, 2016-12-20:
I do not think Newlib has the C++ standard library, so while you can write C++-code, using things that are not a part of the language itself but defined in the standard library (eg stringstreams) will not work.

igrr, 2016-12-20:
This is in pipeline... I need to rebuild libstdc++ with the right settings.

imxieyi, 2016-12-20:
Ok. I will try to avoid using standard C++ libraries as a workaround.

closed 2016-12-20 by imxieyi

chenkaiyao, 2016-12-26:
@igrr how can i get the libstdc++

igrr, 2017-01-20:
This issue (linking against libstdc++ and missing wcsftime) is now fixed in master.