espressif/esp-idf #176: Provide a <netinet/in.h> for TCP/IP

nkolban, 2016-12-20:
I am finding that as I port sockets based libraries over to the ESP-IDF framework they are porting extremely well.  In the majority of cases, no changes at all seem to be needed for them to compile cleanly and execute as desired.   However, there does appear to be at least one common pattern that I am finding that may be able to be easily resolved.

I am finding many references in 3rd party C source to the include:

```
#include <netinet/in.h>
```

Whatever is being included here in Linux is being pulled in from "somewhere else" by the inclusion of other LWIP headers.  However, because this is not present in the ESP-IDF, we can't compile without explicitly removing its reference from the source files being compiled.

This issue asks for some examination of this file (see: http://pubs.opengroup.org/onlinepubs/7908799/xns/netinetin.h.html).

Maybe we can supply it as a "no-op" (an empty include file) in the ESP-IDF framework?  Maybe the framework needs modified to actually include a real version and refactor some of our existing headers to be present in this defined header?

projectgus, 2016-12-22:
I think making a header that includes the appropriate LWIP headers to function would be a good solution for this, similar to what the netinet/in.h header does on Linux.

Is this something you'd be able to add to the lwip component and send as a PR?

nkolban, 2016-12-22:
I did some digging ... looking at this source file ... 

http://git.savannah.gnu.org/cgit/lwip.git/tree/src/include/lwip/inet.h

It seems that the LWIP implementation has made the conscious choice to deviate from a "Linux" like file structure ... at the start of the file we see:

```
/**
 * @file
 * This file (together with sockets.h) aims to provide structs and functions from
 * - arpa/inet.h
 * - netinet/in.h
 *
 */
```

As such, this causes me to think that one should not think of LWIP and Linux kernel level TCP/IP as being the "same" APIs.  I am starting to think that source code portability between LWIP and Linux at the networking level wasn't meant as a primary design choice.

projectgus, 2016-12-30:
> I am starting to think that source code portability between LWIP and Linux at the networking level wasn't meant as a primary design choice.

I think it's still OK to add the wrapper header. It is true (I expect) that for some applications you won't be able to drop code writen for Linux or another BSD Socket interface and have it work without modifications.

As it happens, arpa/inet.h and netinet/in.h headers are being added to esp-idf's lwip component for compatibility with another library. Will let you know once this is merged.

igrr, 2017-01-16:
Available in master since d0b10ba2.

closed 2017-01-16 by igrr