espressif/esp-idf #177: WiFi STA not reconnecting

rojer, 2016-12-21:
Usually ESP WiFi keeps trying to connect to the configured AP unless explicitly `disconnect()`.
This is a nice feature, except sometimes it doesn't.
In this case, it did not lose the connection, this is the first attempt after boot:
```
miot_wifi_set_mode   WiFi mode: STA
I (4401) wifi: frc2_timer_task_hdl:3ffcb65c, prio:22, stack:2048
I (4411) wifi: pp_task_hdl : 3ffcb80c, prio:23, stack:8192
miot_wifi_setup_sta  WiFi STA IP: 192.168.11.9/255.255.255.0 gw 192.168.11.1
I (4421) wifi: mode : sta (24:0a:c4:00:31:bc)
event_handler        event: 2
miot_wifi_setup_sta  WiFi STA: Connecting to Over_9000_Internets
miot_sys_config_init_http HTTP server started on [80]
mg_rpc_channel_uart  0x3ffcfc40 UART0
mg_rpc_add_channel   '' 0x3ffcfc40 uart 1
miot_i2c_create      I2C initialized (SDA: 22, SCL: 23)
miot_init            Init done, RAM: 164392 free, 164392 min free
I (4851) wifi: n:3 0, o:1 0, ap:255 255, sta:3 0, prof:1
I (5491) wifi: state: init -> auth (b0)
I (6491) wifi: state: auth -> init (2)
wifi_event_handler   WiFi STA: disconnected, reason 2
```
the stated reason is `WIFI_REASON_AUTH_EXPIRE`. i'm not sure what it means and if it's meant to be a permanent, but i see no further connection attempts. is this the intended behavior?

rojer, 2016-12-21:
for reference, successful connection attempt to the same AP (after pressing reset):
```
I (4893) wifi: n:3 0, o:1 0, ap:255 255, sta:3 0, prof:1
I (5533) wifi: state: init -> auth (b0)
I (5533) wifi: state: auth -> assoc (0)
I (5543) wifi: state: assoc -> run (10)
I (5563) wifi: connected with Over_9000_Internets, channel 3
I (5563) event: ip: 192.168.11.9, mask: 255.255.255.0, gw: 192.168.11.1
```

negativekelvin, 2016-12-21:
It looks like the auth attempt timed out after 1 second but not sure why no retries.  Whereas the successful  attempt connected in 30ms.

projectgus, 2016-12-22:
Automatic re-association can be achieved via the esp-idf event handler. When you receive `SYSTEM_EVENT_STA_DISCONNECTED`, call `esp_wifi_connect()`

You can see this in some of the examples, ie:
https://github.com/espressif/esp-idf/blob/master/examples/04_https_request/main/https_request_main.c#L145

I marked this as a workaround in the comment when I wrote it, but I now understand that it's the expected behaviour.

rojer, 2016-12-22:
so this is WAI then? sure, that's fine, i'll modify my code to invoke `esp_wifi_connect()` then.

closed 2016-12-22 by rojer