espressif/esp-idf #178: Working SDIO support?

mseerig, 2016-12-21:
I'am working for hours on my problem to interface a SD card to my ESP-WROOM-32.
Is there any example code, or documentation how to use SDIO on ESP32?
I didn't found official, or unofficial documentation. Sarching throw the whole repository, including and all *.h and *.c files gave me also no result. 
Please help me, my new love "ESP32" is in danger.. : (

negativekelvin, 2016-12-22:
Not yet, supposed to be released in the next few weeks.  But you can try spi interface for now.

projectgus, 2016-12-22:
@negativekelvin is correct, a driver for SDMMC support is being worked on right now. A generic SPI driver (which can access SD cards in SPI mode) is also currently in progress.

igrr, 2017-01-16:
Available in master since edd924f.

closed 2017-01-16 by igrr