espressif/esp-idf #179: Non-Volatile Storage

SirRibeiro, 2016-12-21:
Hello i'm using the Non-Volatile Storage example to write some data in the ESP32, everything works fine (great job on that). But i can only write around 90 000 chars on the memory. 
(The way that i come up with this number was writing 100 keys with 90 chars of data, and after that i get the error of full memory)

Is there any way of increasing this space?

Or i'm i doing something wrong?

PS: I've been checking the SPIFFS and VFS but I have not been able to get this working

Thanks

igrr, 2016-12-21:
This number looks right for the default nvs partition size. You can increase it if you use custom partition table, please check relevant documentation here: http://esp-idf.readthedocs.io/en/latest/partition-tables.html

SirRibeiro, 2016-12-21:
Ok, that is perfect, i've now one more question.

I'm using the arduino IDE to program the ESP32, in parallel with some of the esp-idf examples. 

Do you have any idea who to do that in there?
Maybe in sdkconfig.h?

SirRibeiro, 2016-12-21:
Now i see there is a partitions_singleapp.bin, in the espressif\esp32 folder, is that the default bin, that arduino uses?

If so, can i just creat another one with my custom partition and replace that one?

Thanks

projectgus, 2016-12-22:
I believe you should be able to. You can generate the partitions.bin file from the esp-idf build system (or use the gen_esp32part.py tool directory).

@me-no-dev may have some more insight into Arduino's use of the espidf partition table.

me-no-dev, 2016-12-22:
@SirRibeiro yes you can surely create another bin using IDF and then replace the one in Arduino.
As soon as I get to partitions, I will offer different options through the IDE, but for now that is your best option. Or use Arduino as IDF component :)

SirRibeiro, 2016-12-22:
Thank you very much for your help, 
indeed it is the way, it worked like a charm!

I just created my custom partition using the gen_esp32part.py, and then replace the Arduino\hardware\espressif\esp32\tools\sdk\bin\partitions_singleapp.bin

Now i've alot more space to store data.

You all are doing a great job on this.
Merry Christmas

closed 2017-01-15 by igrr