espressif/esp-idf #180: Exception at lwip/core/tcp.c:1953

nkolban, 2016-12-22:
I am hunting down a problem in my application and finding that we crash in LWIP logic.  Here is the backtrace as reported by gdb:

```
Remote debugging using /dev/ttyUSB0
tcp_netif_ipv4_addr_changed (old_addr=0x3fff04c0, new_addr=0x3ffdfc20)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/tcp.c:1953
1953	      if (!IP_IS_V6_VAL(lpcb->local_ip)) {
(gdb) bt
#0  tcp_netif_ipv4_addr_changed (old_addr=0x3fff04c0, new_addr=0x3ffdfc20)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/tcp.c:1953
#1  0x400ea341 in netif_set_ipaddr (netif=0x3fff04bc, ipaddr=0x3ffdfc20)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/netif.c:452
#2  0x400ea382 in netif_set_addr (netif=0x3fff04bc, ipaddr=0x3ffdfc20, netmask=0x3ffc5a10, 
    gw=0x3ffc5a14) at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/netif.c:330
#3  0x400ee3e1 in dhcp_bind (netif=0x3fff04bc)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/ipv4/dhcp.c:1140
#4  0x400eed18 in dhcp_timeout (netif=0x3fff04bc)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/ipv4/dhcp.c:521
#5  0x400eed65 in dhcp_fine_tmr ()
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/ipv4/dhcp.c:476
#6  0x400ea6c6 in dhcp_timer_fine (arg=0x0)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/timers.c:178
#7  0x400ea74c in sys_timeouts_mbox_fetch (mbox=0x3ffbd750 <mbox>, msg=0x3ffc5ad0)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/core/timers.c:575
#8  0x400ed12f in tcpip_thread (arg=<optimized out>)
    at /home/kolban/esp32/esptest/esp-idf/components/lwip/api/tcpip.c:101
(gdb) 

```

I have nothing as yet further to add but am raising this issue in the hopes that if others search, it may show a pattern of issues/solutions.

negativekelvin, 2016-12-22:
Where is the register dump and exception message?  Have you tried enabling stack overflow check in menuconfig?

nkolban, 2016-12-22:
My Component Config > FreeRTOS > Check for stack overflow is set to default which is "check by stack pointer value".

Here is the register dump an exception message ... if you have any knowledge on how to interpret this stuff I'd be very grateful to learn from you ...
```
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400f1229  PS      : 0x00060330  A0      : 0x800ea341  A1      : 0x3ffc5a30  
A2      : 0x3fff0518  A3      : 0x3ffdbb50  A4      : 0x00060523  A5      : 0xb33f0000  
A6      : 0xb33fffff  A7      : 0x3ffc5b30  A8      : 0x00000004  A9      : 0x00000004  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffbd650  
A14     : 0x00000000  A15     : 0xffffffff  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000030  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400f1229:0x3ffc5a30 0x400ea341:0x3ffc5a50 0x400ea382:0x3ffc5a70 0x400ee3e1:0x3ffc5a90 0x400eed18:0x3ffc5ac0 0x400eed65:0x3ffc5ae0 0x400ea6c6:0x3ffc5b00 0x400ea74c:0x3ffc5b20 0x400ed12f:0x3ffc5b50

              Entering gdb stub now.

```

projectgus, 2016-12-22:
> The issue appears to be currently very recreatable.

Can you provide steps to reproduce? If you're unable to publish code publically, would you be able to email me directly? Thanks.

nkolban, 2016-12-22:
The code for reproduction is embedded with a HUGE amount of code now ... however I "solved" the cause of the puzzle ... in that I found out what I was doing wrong ... but it doesn't really sit well with me the result.

At a high level, I have a primary FreeRTOS task that does 95% of the work.  This includes setting up the ESP32 as a station to an access point.  However, for debugging purposes, I also have a second FreeRTOS task that sets itself up as a socket server.  By that I mean pseudo code like the following:

```
socket()
bind()
listen()
while(1) {
   accept()
   do something
   close client ...
}
```

What I had accidentally done was started that task BEFORE setting up the WiFi connection.  Apparently attempting to do those sockets APIs with no WiFi connection didn't cause any logging at those APIs and they apparently "ran" and did "something".   However, while that task was running in the background, the primary task which initializes WiFi then ran and attempted to create my WiFi connection.   When the remote DHCP server returned an IP address for my ESP32 to use ... it appears to be at THAT point that the exceptions described previously occur.

So the core of this puzzle is "What (if anything) does it mean to try and use sockets APIs to form a listener before forming a WiFi connection?   Should those APIs fail? (they apparently didn't)  Do we consider this a "garbage in/garbage out"?  Do we have to "bullet proof" getting into the state of trying to use sockets before we have WiFi?   I honestly don't know ... but as we see ... the failures were introduced by mistakes in my logic ... but the mistakes were not coding errors (i.e. all the bits I did were correct ... just in the wrong order/timing).

If I were sitting at your desk ... I'd probably say "We have bigger fish to fry" ... and you won't find any objection from me.  I'll leave it for you to decide whether we pursue this or not any further.  But at least now we have a trail to examine ... if we see this exception pattern again in the future, we can ask "Are there any existing socket listeners currently running while WiFi is changing?".

nkolban, 2016-12-22:
Closed for just now ...

closed 2016-12-22 by nkolban