espressif/esp-idf #182: Can't make 06_sntp example work

krzychb, 2016-12-22:


igrr, 2016-12-22:
I forgot to update the example and docs, SLOW_MEM has to be enabled if RTC_DATA_ATTR is used.

krzychb, 2016-12-22:
OK, no worries. You have provided leads in ESP32 Programming Guide > [Deep Sleep](http://esp-idf.readthedocs.io/en/latest/api/deep_sleep.html?highlight=slow%20memory#power-down-of-rtc-peripherals-and-memories)

After adding:
```c
esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_ON);
```
example works as expected retaining boot count and current time:

```
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:4
load:0x3ffc000c,len:1880
load:0x40078000,len:3408
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034
I (735) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (735) heap_alloc_caps: At 3FFC192C len 0001E6D4 (121 KiB): DRAM
I (740) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (750) heap_alloc_caps: At 4009BEBC len 00004144 (16 KiB): IRAM
I (760) cpu_start: Pro cpu up.
I (766) cpu_start: Starting app cpu, entry point is 0x40080c04
I (0) cpu_start: App cpu up.
I (782) cpu_start: Pro cpu start user code
I (901) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 1
I (902) cpu_start: Starting scheduler on PRO CPU.
I (908) cpu_start: Starting scheduler on APP CPU.
I (908) example: Boot count: 65
I (908) example: The current date/time in New York is: Thu Dec 22 13:14:35 2016
I (918) example: The current date/time in Shanghai is: Fri Dec 23 02:14:35 2016
I (928) example: Entering deep sleep for 10 seconds
```

closed 2017-01-13 by igrr

santanapablo1975, 2017-03-29:
Hi All, 

I am testing the sntp example with the latest branch of ESP-IDF (ESP-IDF v2.0-rc1-364-gcecdfdb)and I got the reset from timer WDT... then teh "boot count" got a reset also.

******************************
deep sleep (20s since last reset, 20s since last boot)
config Timer
config IO
deep sleep

ets Jun  8 2016 00:22:57
rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0x00

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0008,len:8

load:0x3fff0010,len:3464

load:0x40078000,len:7828

load:0x40080000,len:252

entry 0x40080034

I (44) boot: ESP-IDF v2.0-rc1-364-gcecdfdb 2nd stage bootloader

I (45) boot: compile time 14:05:40

I (87) boot: Enabling RNG early entropy source...

*******

please any advice about it?

Regards

Juan

santanapablo1975, 2017-03-29:
Hi All, on my last comment I sent the log for another deep_sleep test, but is the same problem... here the logs with the sntp, here is the same timer WDT reset..

*******************

[0;32mI (1416) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (1422) example: Boot count: 2[0m
[0;32mI (1422) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (1422) example: The current date/time in New York is: Wed Mar 29 15:19:56 2017[0m
[0;32mI (1432) example: The current date/time in Shanghai is: Thu Mar 30 03:19:56 2017[0m
[0;32mI (1432) example: Entering deep sleep for 10 seconds[0m
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3464
load:0x40078000,len:7828
load:0x40080000,len:252
entry 0x40080034
[0;32mI (44) boot: ESP-IDF v2.0-rc1-364-gcecdfdb 2nd stage bootloader[0m
[0;32mI (45) boot: compile time 13:56:01[0m
[0;32mI (85) boot: Enabling RNG early entropy source...[0m
[0;32mI (85) boot: SPI Speed      : 40MHz[0m
[0;32mI (85) boot: SPI Mode       : DIO[0m
[0;32mI (97) boot: SPI Flash Size : 4MB[0m
[0;32mI (109) boot: Partition Table:[0m
[0;32mI (121) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (143) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (167) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (190) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (213) boot: End of partition table[0m
[0;32mI (226) boot: Disabling RNG early entropy source...[0m
[0;32mI (243) boot: Loading app partition at offset 00010000[0m
[0;32mI (972) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) [0m
[0;32mI (973) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x08528 ( 34088) map[0m
[0;32mI (989) boot: segment 2: paddr=0x00028538 vaddr=0x3ffb0000 size=0x024bc (  9404) load[0m
[0;32mI (1019) boot: segment 3: paddr=0x0002a9fc vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (1042) boot: segment 4: paddr=0x0002ae04 vaddr=0x40080400 size=0x1a904 (108804) load[0m
[0;32mI (1120) boot: segment 5: paddr=0x00045710 vaddr=0x400c0000 size=0x00034 (    52) load[0m
[0;32mI (1121) boot: segment 6: paddr=0x0004574c vaddr=0x00000000 size=0x0a8bc ( 43196) [0m
[0;32mI (1138) boot: segment 7: paddr=0x00050010 vaddr=0x400d0018 size=0x3c198 (246168) map[0m
[0;32mI (1164) boot: segment 8: paddr=0x0008c1b0 vaddr=0x50000000 size=0x00004 (     4) load[0m
[0;32mI (1190) cpu_start: Pro cpu up.[0m
[0;32mI (1202) cpu_start: Starting app cpu, entry point is 0x40080be4[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1235) heap_alloc_caps: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1257) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM[0m
[0;32mI (1278) heap_alloc_caps: At 3FFB7568 len 00028A98 (162 KiB): DRAM[0m
[0;32mI (1299) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1320) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1342) heap_alloc_caps: At 4009AD04 len 000052FC (20 KiB): IRAM[0m
[0;32mI (1363) cpu_start: Pro cpu start user code[0m
[0;32mI (1425) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (1431) example: Boot count: 1[0m
[0;32mI (1431) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (1431) example: The current date/time in New York is: Wed Mar 29 15:20:07 2017[0m
[0;32mI (1441) example: The current date/time in Shanghai is: Thu Mar 30 03:20:07 2017[0m
[0;32mI (1441) example: Entering deep sleep for 10 seconds[0m

Regards

Juan

santanapablo1975, 2017-03-29:
Hi All,

I just want to write down that I made the following update

git clone https://github.com/espressif/esp-idf.git esp-idf-v2.0-rc2
cd esp-idf-v2.0-rc2/
git checkout v2.0-rc2
git submodule update --init

and the error is gone! we can still have this issue close! sorry for the false trigger ;)

best regards!

and thank you all

Juan