espressif/esp-idf #189: i2c clock stretching support

mfgeorge, 2016-12-30:
Hello, 

I have determined that there is an error in the i2c protocol that allows it to timeout prematurely. In some i2c sensors (often ones that have a microcontroller in them for data fusion), the feature of i2c clock stretching is utilized by the slave. I have found that with such sensors and the ESP32, the i2c protocol often experiences a timeout before the sensor has sent the data back. This creates a compatibility issue with some i2c sensors. 

See this [i2c-bus article](http://www.i2c-bus.org/i2c-primer/clock-generation-stretching-arbitration/) for more detail or this [sparkfun diagram](https://cdn.sparkfun.com/assets/8/2/8/8/5/51ae0000ce395f331b000000.png) for a look at what i2c clock stretching may look like in a timing diagram. 

i2c clock-stretching is a part of the defined protocol, and not merely an added feature for it. In order for your micro-controller to technically have full i2c capability, it needs to support i2c clock-stretching. 

The main sensor I was using for testing was the BNO055. It contains a micro-controller for fusing the data/ performing calculations. If a calculation is not ready, the clock line is pulled low by the BNO055 (the slave) until it is ready, at which point the clock line is released and allowed to go high again. At this time the master can then resume pulsing the clock signal to get the data out.

I have been using a Pycom LoPy, which contains the ESP32, using the i2c class of micropython. 

My apologies if this is not the appropriate place to put this issue, however I figured the problem seemed to be at a lower level than Pycom's implementation of i2c. Please let me know if this issue belongs elsewhere.

negativekelvin, 2016-12-31:
1. There is no i2c driver in esp-idf yet
2. Pycom firmware seems to be using a software i2c driver
3. See the difference between:
https://github.com/pycom/pycom-micropython/blob/master/extmod/machine_i2c.c#L60
https://github.com/pycom/pycom-micropython/blob/master/esp32/mods/machine_i2c.c#L66

So not sure whether i2c hardware will support clock stretching when it is released but doesn't seem like this issue fits here.

mfgeorge, 2017-01-09:
Thanks @negativekelvin for correcting where I should put this issue. You are completely right. Sorry to clutter up the issues on this repository. Also, the arduino-esp32 issue perfectly shows that clock-stretching support is not present. However I see now that it is not a problem upstream, but one downstream on both implementations. The fix in pycom's extmod seems like it should work. Thank you and sorry for not poking around longer before submitting my issue.

closed 2017-01-09 by mfgeorge