espressif/esp-idf #192: make flash: syntax error "(" unexpected

keith204, 2016-12-31:
After following these steps (https://github.com/espressif/esp-idf/blob/master/docs/linux-setup.rst), I am getting the following error:  

--

pi@raspberrypi2:~/esp/myapp $ make menuconfig
MENUCONFIG

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

**pi@raspberrypi2:~/esp/myapp $ make flash**
CC src/efuse.o
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: **Syntax error: "(" unexpected**
/home/pi/esp/esp-idf/make/component_wrapper.mk:176: recipe for target 'src/efuse.o' failed
make[2]: *** [src/efuse.o] Error 2
/home/pi/esp/esp-idf/make/project.mk:376: recipe for target 'bootloader_support-build' failed
make[1]: *** [bootloader_support-build] Error 2
/home/pi/esp/esp-idf/components/bootloader/Makefile.projbuild:31: recipe for target '/home/pi/esp/myapp/build/bootloader/bootloader.bin' failed
make: *** [/home/pi/esp/myapp/build/bootloader/bootloader.bin] Error 2
pi@raspberrypi2:~/esp/myapp $

projectgus, 2017-01-09:
Hi Keith,

Sorry for the slow reply. Can you please run "make flash V=1" and copy the command line which is being passed to xtensa-esp32-elf-gcc? Thanks.

Angus

projectgus, 2017-02-16:
@keith204 did you make any progress with this issue?

keith204, 2017-02-21:
It may be a while before I get back into this (and currently don't have a unix environment to quickly tinker with).    It seems that nobody else is having this issue.  Can we consider this closed for now, and then if I run into it again, we reopen?

projectgus, 2017-02-21:
OK, thanks. Feel free to reopen if you do.

closed 2017-02-21 by projectgus