espressif/esp-idf #196: system_get_rtc_time()

ESP32DE, 2017-01-04:
hi guys
happy new year!

test the system_get_rtc_time(void) in
time.c ( newlib ) 
https://github.com/espressif/esp-idf/blob/master/components/newlib/time.c#L203
` uint64_t system_get_rtc_time(void)`

in kolban's esp32 book i read the function 
with inlcude esp_system.h

> system_get_rtc_time
> Get the real time clock cycles.
> uint32_t system_get_rtc_time()
> Page 372
> --------------------------------------------------------
> Page 373
> Retrieve the number of real-time clock cycles.  For example, if the value returned by 
> system_rtc_clock_cali_proc() is 5.75µS then if the system_get_rtc_time() returns 
> 100, then 575µS of wall clock time has passed since the clock was started.
> Includes:
> • esp_system.h

`uint32_t system_get_rtc_time(void)`

but there is no function in the system
https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_system.h

same, there is no declare in the newlib/include/time.h
https://github.com/espressif/esp-idf/blob/master/components/newlib/include/time.h

do we miss / do we wrong  here a part?

txs

best wishes
rudi ;-)

forum: 
[http://esp32.com/viewtopic.php?f=2&t=859#p3621]

ESP32DE, 2017-01-04:
> This  function was intentionally removed, 
 > as the functionality is now provided by POSIX gettimeofday function.

http://esp32.com/viewtopic.php?f=2&t=859&p=3626#p3626

nkolban, 2017-01-04:
The book of notes on the ESP32 has been amended to remove the system_get_rtc_time() function reference.

projectgus, 2017-02-16:
Thanks @nkolban !

closed 2017-02-16 by projectgus

vijayprakash1, 2018-10-16:
> The book of notes on the ESP32 has been amended to remove the system_get_rtc_time() function reference.

how can I access the rtc timer of ULP in deep sleep mode?